bmd-studio/stm32-for-vscode #106: Failed to open device LIBUSB_ERROR_NOT_SUPPORTED - No J-Link device found

RootShell-coder, 2022-05-03:
Sorry to take away your precious time. I'm trying to get the printf() function to work through the debugger console, but I can't get "Hello" from the program. Debug starts and I can perform step -by -step debugging. I can’t flash the microcontroller with a button "Flash STM32". In the launch.json file is visible my weak attempts to make it work. 

I have not made any changes in Makefile and STM32Make.make.
 
```c
#include "stdio.h"
    printf("Hello\n\r"); //ST-LINK v2 SWO debug no job? fuc...oO.la.la
```

launch.json
```json
{
    "version": "0.2.0",
    "configurations": [
        {
            "showDevDebugOutput": "parsed",
            "type": "cortex-debug",
            "request": "launch",
            "name": "Debug J-Link",
            "cwd": "${workspaceRoot}",
            "executable": "${workspaceRoot}/build/learning_STM32_swarm.elf",
            "server": "JLinkGDBServer",
            "serverpath": "D:\\Develop\\STM32\\Tools\\SEGGER\\JLinkGDBServerCL.exe",
            "servertype": "jlink",
            "device": "STM32F103RC",
            "interface": "swd",
            "serialNumber": "MySerialNumber",
            "svdFile": "${workspaceRoot}/.vscode/STM32F103xx.svd",
            "swoConfig": {
                "enabled": true,
                "cpuFrequency": 72000000,
                "swoFrequency": 2000000,
                "source": "probe",
                "decoders": [
                    { "type": "console", "label": "ITM", "port": 0, "encoding": "ascii" }
                ]
            }
        },

        {
            "showDevDebugOutput": "parsed",
            "cwd": "${workspaceRoot}",
            "executable": "./build/learning_STM32_swarm.elf",
            "name": "Debug STM32",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            "preLaunchTask": "Build STM",
            "device": "stm32f103xe.s",
            "configFiles": [
                "openocd.cfg"
            ],
            //https://github.com/posborne/cmsis-svd
            "svdFile": "${workspaceRoot}/.vscode/STM32F103xx.svd",

            // https://github.com/Marus/cortex-debug
            // Initial support for STMicroelectronic's ST-LINK GDB server (no SWO support yet)
            //"swoConfig": {
            //    "enabled": true,
            //    "cpuFrequency": 72000000,
            //    "swoFrequency": 36000000,
            //    "source": "probe",
            //    "decoders": [ { "type": "console", "label": "SWO", "port": 0, "encoding": "ascii" } ] }
        }
    ]
}
```

Debug
```
Waiting for gdb server to start...[2022-05-03T17:55:21.911Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
"D:\\Develop\\STM32\\Tools\\SEGGER\\JLinkGDBServerCL.exe" -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device STM32F103RC -select usb=779590444
SEGGER J-Link GDB Server V7.64c Command Line Version

JLinkARM.dll V7.64c (DLL compiled Apr 20 2022 16:13:22)

Command line: -singlerun -nogui -if swd -port 50000 -swoport 50001 -telnetport 50002 -device STM32F103RC -select usb=779590444
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     50000
SWO raw output listening port: 50001
Terminal I/O port:             50002
Accept remote connection:      localhost only
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               on
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 STM32F103RC
Target interface:              SWD
Target interface speed:        4000kHz
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Firmware: J-Link STLink V2 compiled Aug 12 2019 10:28:03
Hardware: V1.00
S/N: MySerialNumber
Checking target voltage...
Target voltage: 3.30 V
Listening on TCP/IP port 50000
Connecting to target...
Connected to target
Waiting for GDB connection...Connected to 127.0.0.1
Reading all registers
Read 4 bytes @ address 0x080030B8 (Data = 0xB510B1D3)
Received monitor command: halt
Halting target CPU...
...Target halted (PC = 0x080030B8)
Received monitor command: reset
Resetting target
Downloading 484 bytes @ address 0x08000000
Downloading 15984 bytes @ address 0x080001F0
Downloading 5224 bytes @ address 0x08004060
Downloading 416 bytes @ address 0x080054C8
Downloading 8 bytes @ address 0x08005668
Downloading 4 bytes @ address 0x08005670
Downloading 4 bytes @ address 0x08005674
Downloading 116 bytes @ address 0x08005678
Comparing flash   [....................] Done.
```

Flash
```
"D:/Develop/STM32/Tools/xpack-openocd-0.11.0-4/bin/openocd.exe" -f ./openocd.cfg -c "program build/learning_STM32_swarm.elf verify reset exit"
xPack OpenOCD x86_64 Open On-Chip Debugger 0.11.0+dev (2022-03-25-17:32)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : DEPRECATED target event trace-config; use TPIU events {pre,post}-{enable,disable}
Warn : Failed to open device: LIBUSB_ERROR_NOT_SUPPORTED.
Error: No J-Link device found
** OpenOCD init failed **
shutdown command invoked

make: *** [STM32Make.make:245: flash] Error 1
Процесс терминала "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command & 'D:/Develop/STM32/Tools/xpack-windows-build-tools-4.3.0-1/bin/make.exe' -j16 -f STM32Make.make  flash" был завершен с кодом выхода 1.
```

jortbmd, 2022-05-03:
Hi! Thank you for your consideration but no worries. 
So there seem to be a few things going on. One is that you switched from st-link to j-link. Which is possible however in the second part it seems that openocd had an issue with opening the J-Link debugger/device. As far as I can tell the flashing using j-link and the j-link gdb server worked. I would advise to use this instead of openocd if you already have it installed. 
E.g.:
```json
{
            "showDevDebugOutput": "parsed",
            "cwd": "${workspaceRoot}",
            "executable": "./build/learning_STM32_swarm.elf",
            "name": "Debug STM32 J-Link",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "jlink",
            "preLaunchTask": "Build STM",
            "device": "stm32f103xe.s",
            "configFiles": [
                "openocd.cfg"
            ]
}
```
And set the "cortex-debug.JLinkGDBServerPath" in your settings.json.

The other thing is printf over debug. I have experience with st-link and openocd, if you want to achieve this you need to enable something called semihosting. A good explanation can be found here: https://shawnhymel.com/1840/how-to-use-semihosting-with-stm32/
I haven't delved into this for J-Link but it might be different.

For the STM32-for-VSCode variant of this is by changing the linkerFlags and libraries in the STM32-for-VSCode.config.yaml e.g.:
```yaml
linkerFlags:
  - -specs=nano.specs
  - -specs=rdimon.specs

# libraries to be included. The -l prefix to the library will be automatically added.
libraries:
  - c
  - m
  - nosys
  - rdimon
```

Add these lines to the end of  openocd.cfg file:
```
init # initialize OpenOCD
arm semihosting enable #enable semihosting.
```
E.g.:
```
#OpenOCD configuration file, generated by STM32 for VSCode

# Programmer, can be changed to several interfaces
# Standard will be the stlink interface as this is the standard for STM32 dev boards
source [find interface/stlink.cfg]

# The target MCU. This should match your board
source [find target/stm32f1x.cfg]

# Enable semihosting
init
arm semihosting enable
```

And finally define the  ``extern void initialise_monitor_handles(void);`` somewhere at the top of your main.c file and call the ``initialise_monitor_handles();`` during your setup.
e.g.:
```c
 /* Private user code ---------------------------------------------------------*/
/* USER CODE BEGIN 0 */
extern void initialise_monitor_handles(void);
/* USER CODE END 0 */

/**
 * @brief  The application entry point.
 * @retval int
 */
int main(void) {
  /* USER CODE BEGIN 1 */
  initialise_monitor_handles();
  /* USER CODE END 1 */
  ... more code
```

RootShell-coder, 2022-05-04:
-    _Oh, it seems to me that all this is too confusing. I had a hysteria when I found out that there are a couple of settings.json are not different in name. Why are you so with me, Microsoft?_ 

-    With J-link (JSсope), data output to the console becomes unnecessary. Excellent acquisition. I read anyway "How to Use Semihosting with STM32"
-  I'm still trying to fix Json configuration files. I had to reinstall VScode when I finally got confused. But thanks to your work, it is easier for me to understand how everything is connected. Close the question, I will find it later, when I rest a little, to continue to study the magical world of programming.

_Thank you for your help_

closed 2022-05-04 by RootShell-coder

jortbmd, 2022-05-06:
Hi! Glad I could help. If you get stuck on the configuration just post it here again. It might be a nice googleable resource for others and I can probably help you with it. Good luck with your endeavours!