bmd-studio/stm32-for-vscode #114: Installation OpenOCD not found

emirgo, 2022-08-13:
Getting the below warning during install build tools execution:

```
npm WARN config global `--global`, `--local` are deprecated. Use `--location=global` instead.
@xpack-dev-tools/openocd@0.11.0-4.1...
error: Package @xpack-dev-tools/openocd@0.11.0-4.1 not found
```

Any pointers as to how this could be solved are welcome.

jortbmd, 2022-08-17:
Could you tell me which OS you are using? There is a known issue with 32-bit windows.

emirgo, 2022-08-17:
> Could you tell me which OS you are using? There is a known issue with 32-bit windows.

64-bit windows installation, it's a new laptop device.

jortbmd, 2022-08-17:
Could you try it again today? It might have something to do with it being the latest package. I have just run a test once more and as far as I can tell it should not fail with the same package it was trying to install on your pc.