bmd-studio/stm32-for-vscode #116: Incorrect YAML formatting results in silent delete and rebuild of YAML file

jnz86, 2022-08-19:
I've noticed that if you incorrectly format a YAML entry and build, that the entire file seems to be deleted and rebuilt using the original makefile from the MX system. The issue is this happens silently, and the make process is continued as usual. 

If the "example" entry below has an additional `"` at the end, this will result in all YAML file changes being removed and rebuilt from original make. 

```
IncludeDirectories: 
  - "Core/**"
  - "Drivers/**"
  - "../../../../example/**""
  - "../../../../notice/the/extra/double/quote/above/"
```

If a warning and fixme type of halt can not be shown, is it possible to at least inform the user via a toast that 
YAML FILE ERROR, REBUILT FROM MAKEFILE" or something like that be shown to the user? It took me quite a bit of fiddling to figure out what was happening without notifications.

Thanks!

JohannesRgnr, 2022-08-31:
I had a similar issue today. While editing some source files entries, the YAML file got back to default settings without warning.

jortbmd, 2022-09-05:
Thanks for opening up this issue. I will integrate this in a next release.

jortbmd, 2022-12-08:
This issues has been fixed in the current release (3.2.3). If you find any more issues concerning this feel free to re-open the issue or create a new one.

closed 2022-12-08 by jortbmd