bmd-studio/stm32-for-vscode #117: STM32H745 dual core  support?

dookei, 2022-09-06:
Hi,
 is there support for STM32H745 dual core ? This do not allow to generate makefile under STM32CubeMx

dookei, 2022-09-07:
Ok thanks for your fast feedback

dookei, 2022-09-12:
just reopening the issue.

Andreychik32, 2024-01-28:
It's actually possible now to generate dual-core projects in makefile format.
Waiting for plugin to support dual-core MCU's.

jortbmd, 2024-01-29:
@Andreychik32  thanks for notifying me. Will have a look into this when some time opens up. If ST implemented it for the makefile the integration should become a lot easier to do. Will let you know when I have had time to look into it.

biobuilder, 2025-04-02:
Would really like to use Dual Core support with your extension.  Seems like your extension is much simpler and better than the alternatives.

We've been trying to force the system to configure a dual-core project for STM32WL5MOCH, but unfortunately no success as of yet.