bmd-studio/stm32-for-vscode #125: Fix OpenOCD error when using stm32 LoraWan 1

signorettae, 2022-11-27:
Hi, 
when I created a project for a [STM32 LoRa WAN 1 discovery](https://www.st.com/en/evaluation-tools/b-l072z-lrwan1.html) I got an error during the upload of the firmware.
The target MCU in the `openocd.cfg` file was set as it follows:
`source [find target/stm32l0x.cfg]`
while it shoud be
`source [find target/stm32l0.cfg]`.

It could be a fix for a future release.
Please let me know if you need more info.

Bye,
Emanuele

jortbmd, 2022-11-29:
Hi! Thanks for opening up an issue. You are absolutely right on this. Somehow the l0 and l1 series do not follow the same format as the rest. I see you already created a pull request for this. Thank you for this! The pull request seems good and will merge soon and include it in the next release.

signorettae, 2022-11-29:
Thank you so much!

closed 2022-11-29 by signorettae