bmd-studio/stm32-for-vscode #127: Unable to match requested speed 8000 kHz, using 4000 kHz

CPowa, 2022-11-28:
Hello. I'm a noob at this whole workflow.
Everything seems fine when I build my project but when I try to flash/ debug I get this log

`Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 2000 kHz
Info : STLINK V2J39M27 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.256131
Info : [stm32f4x.cpu] Cortex-M4 r0p1 processor detected
Info : [stm32f4x.cpu] target has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f4x.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08001bd0 msp: 0x20020000
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
** Programming Started **
Info : device id = 0x10006421
Info : flash size = 512 KiB
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
shutdown command invoked`
Any Ideas how to solve it?

CPowa, 2022-11-28:
Comment with actual log (easier to see this way)
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : clock speed 2000 kHz
Info : STLINK V2J39M27 (API v2) VID:PID 0483:374B
Info : Target voltage: 3.256131
Info : [stm32f4x.cpu] Cortex-M4 r0p1 processor detected
Info : [stm32f4x.cpu] target has 6 breakpoints, 4 watchpoints
Info : starting gdb server for stm32f4x.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x08001bd0 msp: 0x20020000
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
Info : Unable to match requested speed 8000 kHz, using 4000 kHz
** Programming Started **
Info : device id = 0x10006421
Info : flash size = 512 KiB
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
shutdown command invoked

jortbmd, 2022-11-29:
Hi! Thanks for opening an issue. As far as I can tell this should be totally fine. The log is not the most clear or friendly but it seems that you successfully uploaded the firmware to your device.
Are you having any issues with debugging?

CPowa, 2022-12-08:
Yes. I can't start debugging due this error in VS Code. It exits right after this log is shown. If you would like me to send you other infos, please let me know.
Sorry for late response but we had a lot of holydays one after another.

jortbmd, 2022-12-08:
No worries. I am not the one having the issue ;).
The log you showed me is there to upload the firmware to the device. After this it should start the debugging session. The fact that it uploads tells me that the debugging tools are installed correctly and that your device is connected properly.
Could you tell me how you start your debugging session? Do you go to the bugging panel and start Debug STM32?
Could you also provide me your launch.json and tasks.json files which are located in the .vscode folder?

jortbmd, 2023-03-28:
No problem. Good to hear that your issue has been resolved. Seeing as it is resolved I will close the issue. Should you have any issues in the future feel free to open up a new one.

closed 2023-03-28 by jortbmd