bmd-studio/stm32-for-vscode #129: MAKEFILE variables

davidstonely, 2022-12-04:
Hello, very nice extension!!!

I am interested in getting git hash variable defined in my builds as described here

[https://stackoverflow.com/questions/1704907/how-can-i-get-my-c-code-to-automatically-print-out-its-git-version-hash](https://stackoverflow.com/questions/1704907/how-can-i-get-my-c-code-to-automatically-print-out-its-git-version-hash))

I couldn't figure out how to edit the STM32-for-VSCode.config.yaml file so that the GIT_VERSION variable is defined in the makefile. I was able to get it to work correctly if I directly edit the makefile, but it gets overwritten easily that way.

jortbmd, 2022-12-05:
Thank you!
This would be nice. I do not have a nice wat as of yet to include it. You could potentially create a header file with the define in it. Using an additional make rule and create a new header file with the definition. I would recommend a header file because if you use a -D flag it will recompile your entire project each time that flag changes.

For example if you would add this to you STM32-for-VSCode.config.yaml:
```yaml
customMakefileRules:
- command: gitversion
  rule: $(shell git describe --abbrev=4 --dirty --always --tags | sed 's/.*/#pragma once \n#define \"&\"/' > Core/Inc/git_version.h)
```
you can run:  ``make -f STM32Make.make gitversion`` to create a git_version.h file in Core/Inc
You can also add this to the tasks.json in vscode and have the Build STM task depend on it, so it will be re-generated each time you run build.

davidstonely, 2022-12-05:
The suggestion of autogenerating a header file is in the stackoverflow posts. The method usually has the overhead of requiring additional build dependencies, such as perl, awk or sed as in your suggestion.   It would nice to have the option to define the variable in the makefile.

jortbmd, 2022-12-08:
Another option is to have an .env file which has the definitions in them and add it to the cDefinitionsFile in the STM32-for-VSCode.config.yaml. And autogenerate that file somehow. 

I can understand that it might be convenient to add a system like this for additional makefile configurations. I have to think about this, but it might be a nice feature to add.

BlazejKrysztofiak, 2024-06-13:
Hi
I am trying to do a similar thing and I added a command to task.json but the problem is it is not run when I click Build from extension tab, I mean:
[image]

But it is run when I build from tasks.
[image]

How can I make it to run the command when building using first way?