bmd-studio/stm32-for-vscode #130: Hex flags

AndreyShi, 2022-12-12:
i did not find how add additional hex flags ? I want to include more flags besides those that exist (-O ihex) for .hex file

jortbmd, 2022-12-19:
Hi thanks for opening an issue for this. As of yet there is not a direct way to add flags for the generation of the .hex file specifically.
Which flags would you like to add?
This might be an easy feature to add so I might incorporate it in a next release.

AndreyShi, 2023-01-13:
in my project i have custom sections which i want delete in hex output ,example: arm-none-eabi-objcopy.exe -R .settings -R .factory -O ihex. And i need some operations for binary output, example: arm-none-eabi-objcopy -R .factory -R .settings -R .RAMVectorTable --pad-to 0x0801d000 --gap-fill=0xFF -O binary, so I would so like to be able to add my flags for HEX and BIN.
And more , i think that good idea is make  pre build steps and post build steps. Kind regards.

jortbmd, 2023-03-28:
My apologies for not getting back to you sooner. Yo might be able to do this by using the customMakefileRules section in the STM32-for-vscode.config.yaml file.
e.g.:
```
# Custom makefile rules
# Here custom makefile rules can be added to the STM32Make.make file
# an example of how this can be used is commented out below.
customMakefileRules:
 - command: hexout
   rule: arm-none-eabi-objcopy.exe -R .settings -R .factory -O ihex
   dependsOn: $(BUILD_DIR)/$(TARGET).elf # can be left out
 - command: binout
   rule: arm-none-eabi-objcopy -R .factory -R .settings -R .RAMVectorTable --pad-to 0x0801d000 --gap-fill=0xFF -O binary
   dependsOn: $(BUILD_DIR)/$(TARGET).elf # can be left out
```
Would this be sufficient for your use case?

jortbmd, 2025-01-20:
Closing this issue due to inactivity. If an issue arises again or if the same issue persist feel free to re-open this issue or create a new issue.

closed 2025-01-20 by jortbmd