bmd-studio/stm32-for-vscode #131: After 'Install Build Tools', gcc is still not installed

Strooom, 2022-12-18:
I'm trying to get this working on Win11.
The extension complains about 'one or more build tools have not been found'.
They are all installed on my machine, as I'm using PlatformIO regularly. So I pointed to the path, with no success, then I pointed to the .exe, with no success either : the extension keeps complaining about build tools not being there.
So then I decided to let it install the tools itself. After a few minutes, I see that Make and OpenOCD have been installed, but gcc is not, ie. there is a folder created next to the other 2 tools, but there is no bin folder there with the actual executables.

Running the install again reveals that the tools should be installed..
[image]
and there is a popup with this error
[image]

jortbmd, 2022-12-19:
Thanks for opening up a new issue for this!
Could you perhaps try to delete the arm none eabi folder. It should be in: c:\Users\__YOUR_USER_NAME\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\@xpack-dev-tools
And then re-install. It might be that the first install failed and due to that that it thinks everything is already there and does not install the arm toolchain fully. 

If the issue persists please let me know. I would also like to know which shell you are using e.g. fish, powershell, cmd and which version of STM32 for VSCode you have installed.

Strooom, 2022-12-19:
Mmm, I am trying, but now I can't even get the Extension to appear in the sidebar.. (Why do you hide the extension, unless you are in an STM32 project ? I don't know of any other VSC extensions that behave like this)

Here is what I did : 
* install the extension from the Marketplace. Its shown in the installed extension.
* then I made a test project in STM32CubeMx, with makefile option and saved it.
* then I open the folder for this project in VSCode, the files are shown, but the extension does not show itself in the sidebar

[image]

From the screenshot you can see stm32-for-vscode V3.2.3 is installed, 
you see the root of my project with .ioc file, .mxproject file, etc.
You see extensions such as PlatformIO, ToDo and GitLens, but stm32-for-vscode is not shown.

jortbmd, 2022-12-20:
It should open automatically when it finds an .ioc file. So it is weird that it would not open for this project.
The hiding of the extension is done on purpose. VSCode is used for a lot of different kinds of programming and I did not want STM32-for-VSCode hogging boot time when VSCode is being used for something else then programming STM32.
For now there is a command which you can use STM32: Open the STM32 for VSCode extension. You can open the command menu by pressing ctrl+shift+p. There are also commands for installing build tools and checking for the build tools. ("STM32: Install all the build tools for the STM32 for VSCode extension" and "STM32: Check if the required build tools are present for STM32 for VSCode"

Strooom, 2022-12-20:
Ok Jort, I am making progress:
* deleted the folder with build tools and retried to install them, this time it worked fine.
* then I tried to build the project, this also seemed to work - getting a number of warnings but I think they are all about some files missing in the project, not about the extension. 
* the extension then created a .yaml file in the root of the project, I was hoping this would be recognized and the extension would show in the sidebar, but still no success. I can only access it from the command (ctrl+shift+p)

As for your remark why the extension is only visible when opening a relevant project : I understand, and agree. But at the same time, to disable/enable the extension is only 2 clicks away, and I think a more standard way to handle this.

I will close the issue now, thanks for your quick support and 'Prettige eindejaarsfeesten'.

closed 2022-12-20 by Strooom