bmd-studio/stm32-for-vscode #132: Undefined reference when building touchGFX projects

madsonrl, 2022-12-19:
Hello,

I'm trying to build and flash stm32-projects, including touchGFX.
I can build any stm32-project created as makefile-project from cubeMX, but when adding touchGFX it gives error with undefined reference to `BSP_QSPI_Init', `videoTaskFunc(void*)', `IdleTaskHook', `touchgfx_init' and `touchgfx_taskEntry' . I reassembled my main.c file to main.cpp and added the touchgfx includeDirectories and sourcesfiles to the STM32-for-VSCode.config.yaml file, but I must be missing something.
Thank you if you can help.

Madson

jortbmd, 2022-12-20:
Hi Madson,

Thanks for opening up an issue. It seems that there are still some files missing. As far as I can tell you need to have the BSP (Board support package) for your specific board and probably some generated functions/tasks from touchGFX that are now not present.
If you could share your project I can have a look for you as TouchGFX support is still one of the features I want to add.

madsonrl, 2022-12-20:
Hi,  thanks for your help. 
The BSP files are in the TouchGFX folder. Here is the public repository of this project so you can take a look.
[https://github.com/madsonrl/TEST_IHM_FW_VSCode](url)

jortbmd, 2023-01-12:
Just as an update I have tried to make it work for a while however it took me a bit too much time. Will still keep this on my todo list, but might be a bit longer until you hear from me. If you are successful in the mean time please let me know.