bmd-studio/stm32-for-vscode #134: Download code to external flash

matchrom, 2023-01-10:
Hi,

As I know, this extension can't download and debug code to an external flash, of course with a correct stldr file. Do we have a plan to develop this function?

Thanks for the great work, keep going.

jortbmd, 2023-01-12:
Hi!
As of yet external flash support is not on the roadmap. Could you perhaps explain a bit more what your specific use case is. This way I can tell a bit better how much work it is and if it makes sense to put it on the roadmap. Is this a specific STM chip which only uses external flash? Or are you aiming for tighter integration of the debug and upload experience for your current set-up and if so what kind of set-up do you have. Also do you have any example IDE's which do implement this functionality natively?

jortbmd, 2023-01-24:
Thanks for your response. I will look into this once I have a bit of time available. Could you also perhaps share the stldr file as a reference?

matchrom, 2023-01-28:
Hi，you can find stldr files in 'STM32CubeProgrammer Install Path\bin\ExternalLoader'. 

And also, you can find flashloaders source code of external memories [here](https://github.com/STMicroelectronics/stm32-external-loader).

jortbmd, 2023-01-31:
Thanks! Will look into it once I have the chance. I will update you once I have looked into it.

jortbmd, 2023-03-20:
That is weird. If you could share the project or a project with similar behaviour I can check and debug that.

For the solution that sounds great! This might be a good way to do it in general. Can I ask from which extension you are getting the ``command:vscode-embedded.st.cubeprogrammer``? Seems it might be worth looking into as it has some dependencies that might be convenient for use with STM32 for VSCode.

AndrewCapon, 2023-03-20:
Hi @jortbmd 

Well It seems I am a total fool as I have been posting here thinking this was repo for https://marketplace.visualstudio.com/items?itemName=stmicroelectronics.stm32-vscode-extension

I was searching for a similar issue and thought this was the right place!

I'm really sorry!

Andy

jortbmd, 2023-03-20:
@AndrewCapon  Haha no worries. This is the first time I have seen this extension. And this would have probably been the extension I wished I would have seen when starting with my own. So I am going to have a real in depth look at this ;).

jortbmd, 2023-03-28:
Closing this issue for now. However feel free to let me know how you experience the official ST extension. For now I have a larger user base then ST and would be a nice challenge to see if I can compete a bit with them. So if you could let me know what you love or hate that would be great ;).

AndrewCapon, 2023-03-28:
Absolutely, sorry I meant to answer your last post and then got waylaid!

The ST extension as it stands for me is not fit for purpose yet, too many issues.

I guess I should look at yours :)