bmd-studio/stm32-for-vscode #136: Cant get stm32-for-vscode to work

mjerjawi, 2023-01-23:
I have installed the extension according to the description but still stuck with the "One or more tools have not been found" what I'm missing ?
[image]

jortbmd, 2023-01-24:
Hi! Thanks for opening an issue for this.

Looks like all tools have been installed. It might be that it failed half way through so that the folders are there but not the actual executables. If you delete everything in C:\Users\mjerj\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode this might solve your issue. If not click the check build tools after installation. If that does not work please let me know, because then it might be a different issue.

mjerjawi, 2023-01-24:
> Hi! Thanks for opening an issue for this.
>
> Looks like all tools have been installed. It might be that it failed half way through so that the folders are there but not the actual executables. If you [...]

That worked like a charm 
Issue is solved 
Thank you a lot, have a nice day

jortbmd, 2023-01-31:
Good too hear! Glad I could help.

I will close the issue for now. Should you have any further issues in the future do not hesitate to create a new issue.

closed 2023-01-31 by jortbmd