bmd-studio/stm32-for-vscode #137: Feature Request: Add Debug/Release Configuration Options

cvanbeek13, 2023-01-30:
This is an awesome extension to get going with STM32 development in VSCode!  One feature that I'm hoping to see is support for changing the `DEBUG = 1` line in the Makefile and also changing the additional `CFLAGS` and `CXXFLAGS` that are set when `DEBUG` equals `1`.

[image]

I was thinking this would involve adding `cDebugFlags` and `cxxDebugFlags` lists to the yaml file.  Maybe instead of adding control over the `DEBUB = 1` flag to the yaml file, there can be separate build tasks in `tasks.json` for debug and release configurations?

I may have a chance to try to make these changes myself in the future.

jortbmd, 2023-01-31:
Thanks! And thank you for opening up an issue for this. This has been on my to do list for a while. However I want to also integrate it with some way of allowing some different kinds of builds for example for unit testing. I have some dedicated time planned in the near future were I will work on this. I will keep you posted!
If you have any more suggestions for how to approach this feel welcome to chime in. For now I was thinking about making different configuration files for main configuration, debug, release and testing.

cvanbeek13, 2023-01-31:
Great, that sounds like an even better idea.  Getting unit tests working for embedded systems is always a bit of a nightmare.  Thanks for continuing to support this extension!

inflowmini, 2023-03-10:
It would be awesome if this could work like CI/CD files where multiple configs are defined in one file. I.E.

prebuild;
 // specific configs
debug:
// ...
release: 
// ...
test:
// ...
postbuild:
// ...

jortbmd, 2023-03-13:
@cvanbeek13  thanks for the kind words!
@inflowmini  Yes might be nice. As stated in issue #144 I am thinking about the future of the build system of the extension. Currently considering going down a pre-existing route e.g. CMake or expanding on what I have build so far. Any input or opinions on this are more than welcome.

jortbmd, 2023-10-19:
Hi all. Quick little update a beta version for the new release is out which allows for debug and release builds. The implementation for it is still incomplete, as I still need to add the generation of the debug and build tasks to the task.json and launch.json files. However it does work for the most part. You can download the beta over here: https://github.com/bmd-studio/stm32-for-vscode/releases/tag/v3.2.6-betav1.
If you find any issues or have any feedback feel free to let me know.

jortbmd, 2025-01-20:
Just a heads up the new version v3.2.10 allows for debug and release builds. I did see that I did not add specific debug and release flags options in the configuration, so this might be nice to add. So leaving the issue open for now until that has been introduced.