bmd-studio/stm32-for-vscode #140: ST button not showing up

DiegoTheMod, 2023-02-13:
Hi I have just started trying to use the software but I can't get it to recconize my project files and start stm32 for vscode.
[image]

jortbmd, 2023-02-14:
Hi!
Thanks for opening up an issue for this. Could you tell me what kind of files are in your workspace? It generally looks for a .ioc file on start-up. Should that fail there is also a command which you can use (use ctrl/cmd + shift + p to open up the command pop-up). The command is: "STM32: Open the STM32 for VSCode extension".
Hope this helps. Please let me know if the issue persists.

jortbmd, 2023-03-28:
Will close this issue due to inactivity. If the issue persists feel free to re-open up this issue again.

closed 2023-03-28 by jortbmd