bmd-studio/stm32-for-vscode #141: Certificate_expiration

vincent01720, 2023-02-19:
Hi,
I've encountered an issue about a certificate expiration, is there anything from my side I can do to fix it?
[image]

mjerjawi, 2023-02-23:
I had a similar issue seems like your installation failed half way through try to delete everything in C:\Users\[USERNAME]\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode
it may resolve your issue

jortbmd, 2023-02-27:
Hi!
Thanks for opening up an issue. What mjerjawi said might solve the issue as an old installation of node maybe present there which  has an old certificate. If the issue persists please let me know.

jortbmd, 2023-03-28:
Quick update I have updated the installation flow for tools, which should remove any old node installation which could give the above behaviour.
Due to inactivity I will close this issue. Should it persist feel free to re-open up the issue and I can look into it a bit further.

closed 2023-03-28 by jortbmd