bmd-studio/stm32-for-vscode #143: .S files don't get added to ASM_SOURCES

jazzpi, 2023-03-05:
I'm trying to get the AzureRTOS middleware working with this extension. AzureRTOS contains a couple of assembly files:

```sh
$ ls Middlewares/ST/threadx/ports/cortex_m7/gnu/src/
tx_misra.S                   tx_thread_context_save.S       tx_thread_interrupt_disable.S  tx_thread_schedule.S     tx_thread_system_return.S
tx_thread_context_restore.S  tx_thread_interrupt_control.S  tx_thread_interrupt_restore.S  tx_thread_stack_build.S  tx_timer_interrupt.S
```

Even if I explicitly list a file under `sourceFiles` in `STM32-for-VSCode.config.yaml`, it doesn't get added to `ASM_SOURCES` in `STM32Make.make`. If I rename the file to use a lowercase `.s` extension, it gets added correctly.

jortbmd, 2023-03-06:
Hi! Thansk for opening up an issue for this. The current quick fix for this is to use lowercase .s assembly files as you are doing now. Currently the extension will include those. However it should be an easy fix to also include the uppercase .S file extension so I will include it in the next release.

jortbmd, 2023-03-28:
Hi! The new release is out. Capitalised .S assembly files should now be included in the build. This should fix the issue and for this reason I will close the current issue. Should any issue arise in the future feel free to re-open this issue or create a new one.

has213, 2023-03-29:
Hi jortbmd,

Maybe there is an Error in the makefile generation

[image]
[image]

There are no UPPER_CASE_ASM_SOURCES and LOWER_CASE_ASM_SOURCES only ASM_SOURCES, which will not be included in OBJECTS

jortbmd, 2023-03-29:
Hi! thanks for opening up this issue. This is correct somehow this snuck through. Currently working on a fix and will publish it ASAP. The issue is currently tracked using issue #150. Will update that issue once it has been resolved.