bmd-studio/stm32-for-vscode #144: Add Ability To Change output/build directory

inflowmini, 2023-03-09:
Currently, the output folder is "build". It would be nice to be able to change it in the yaml file and then have it update the launch file automatically.

jortbmd, 2023-03-10:
Hi thanks for opening up an issue for a feature request!
Might be an interesting feature but it does have some implications for the build system. Can I ask why you want to change the build folder?

jortbmd, 2023-03-13:
Thanks for the clear explanation. This makes a lot of sense and I would the extension to be able to co-exist with other type of build systems and IDE's.
I will think about this. I am currently considering options for the build system and configuration of STM32 for VSCode. Currently due to continual feature addition I kind of created a very limited build system. I am currently considering of going down the route of expanding this or using something like CMake.
Should you have any ideas about his feel free to share as I am open for input.

jortbmd, 2023-10-19:
Hi. I have a little update about this. I have released a beta version of an incomplete release, which would allow you to change the build directory. In the new version a .stm32env environment file is generated and it is possible to add a build variable name e.g.
```
BUILD_DIRECTORY=other_build_directory
```
This should change the build directory while building. If you want to try it you can find the beta release here: https://github.com/bmd-studio/stm32-for-vscode/releases/tag/v3.2.6-betav1.
If you have any feedback please feel free to share it.

jortbmd, 2025-01-20:
Closing this issue as the feature has landed in v3.2.10. Should any issues arise feel free to re-open this issue or make a new one.

closed 2025-01-20 by jortbmd