bmd-studio/stm32-for-vscode #146: How to compile main.c as C in C++ build?

ARNik, 2023-03-19:
Hi!
I would like to compile main.c as C, but if I use C++ setting in yaml file it makes a copy of main.c to main.cpp and compiles it.
Can I turn off this behavior?

I use C++ code as well, and run it from main using wrappers and extern "C" linkage.

jortbmd, 2023-03-20:
Hi thanks for opening up an issue! This is an edge case I did not design for, however it might be an easy feature to add with an extra flag. Could you maybe check if it also does not compile if you set the language to C? I do not expect it to but it is worth a shot.

ARNik, 2023-03-20:
That is a good idea!
I changed the setting back to C, and now it build main.c with my C++ extra code.

The main advantage is that I do not need to rename main.c <-> main.cpp back and forth when I regenerate code.

jortbmd, 2023-03-20:
Good to hear! Does this mean that the issue is solved?

ARNik, 2023-03-20:
Yep, I think so.