bmd-studio/stm32-for-vscode #150: ASM_SOURCES not included in OBJECTS

Flo2410, 2023-03-29:
As mentioned by @has213, the ASM_SOURCES are no longer included in the OBJECTS after the last update with the fix for #143.

Adding ``OBJECTS += $(addprefix $(BUILD_DIR)/,$(notdir $(ASM_SOURCES:.s=.o)))`` fixed the issue.

              Hi jortbmd,

Maybe there is an Error in the makefile generation

[image]
[image]

There are no UPPER_CASE_ASM_SOURCES and LOWER_CASE_ASM_SOURCES only ASM_SOURCES, which will not be included in OBJECTS

_Originally posted by @has213 in https://github.com/bmd-studio/stm32-for-vscode/issues/143#issuecomment-1488459723_

m4tto, 2023-03-29:
I have the same issue. The startup script is not compiled and the generated elf file is basically empty.

jortbmd, 2023-03-29:
Thanks for opening up this issue! I am currently on it somehow this got through the whole pipeline without triggering errors. Just tested the fix and will try to resolve this ASAP. Will keep you posted

jortbmd, 2023-03-29:
Just published the new release. Could everybody please checks that this resolves the issue? Also thanks for notifying me this quickly!

Flo2410, 2023-03-29:
Yes, it works.  Thanks!

FYI: The comment ``# list of ASM program objects`` is duplicated.

jortbmd, 2023-03-29:
Great to hear that it is working. @Flo2410 Thanks for pointing out the double comment. For now I will fix this in an upcoming release. 
As it is working now I will close this issue. Should any of you experience a new issue or wants a feature request don't hesitate to open up a new issue.

closed 2023-03-29 by jortbmd