bmd-studio/stm32-for-vscode #152: Unable to build fresh project from CubeMX when exporting using Makefile toolchain. Sorry, Unimplemented error

yamaan93, 2023-04-05:
So, I have been attempting to start a simple new project from CubeMX. According to the readme the correct way to start the project is with the Makefile option in the toolchain selector in CubeMX. However, when I attempt to build the project, I get the following terminal output error. 

```
Drivers/CMSIS/Include/cmsis_gcc.h: In function '__get_CONTROL':
Drivers/CMSIS/Include/cmsis_gcc.h:219:1: sorry, unimplemented: Thumb-1 'hard-float' VFP ABI
  219 | "c /Users/Yamaan/App{D ta/Roaming/Code/User/globalStorage/bmd.stm32-for-vscode/@xpack-dev-tools/arm-none-eabi-gcc/12.2.1-1.2.1/.content/bin/arm-none-eabi-gcc" -c  -mthumb -mfpu=fpv4-sp-d16 
-mfloat-abi=har
      | d - STM32U575xx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -ICore/Inc -IDrivers/CMSIS/Device/ST/STM32U5xx/Include -IDrivers/CMSIS/Include -IDrivers/STM^32 5xx_HAL
_Driver/Inc -IDrivers/STM32U5xx_HAL_Driver/Inc/Legacy -Og In file included from -W ll -fdata-sections -ffunction-sections -g -gdwarf -ggdb -Wall -fdata-sections -ffunction-secDrivers/CMSIS/Include/cmsis_compiler.h:54 ions  -MMD -MP -MF"build/system_stm32u5xx.d" -Wa,-a,-ad,-alms=build,
                 from /sy tem_stm32u5xx.lst Core/Src/system_stm32u5xx.c -o build/system_stm32u5xx.o
Drivers/CMSIS/Include/core_cm33.h:206,
                 from Drivers/CMSIS/Device/ST/STM32U5xx/Include/stm32u575xx.h:229,
                 from Drivers/CMSIS/Device/ST/STM32U5xx/Include/stm32u5xx.h:103,
                 from Drivers/STM32U5xx_HAL_Driver/Inc/stm32u5xx_hal_def.h:33,
                 from Drivers/STM32U5xx_HAL_Driver/Inc/stm32u5xx_hal_rcc.h:28,
                 from Core/Inc/stm32u5xx_hal_conf.h:265,
                 from Drivers/STM32U5xx_HAL_Driver/Inc/stm32u5xx_hal.h:29,
                 from Core/Inc/main.h:30,
                 from Core/Src/stm32u5xx_hal_msp.c:22:
Drivers/CMSIS/Include/cmsis_gcc.h: In function '__get_CONTROL':
Drivers/CMSIS/Include/cmsis_gcc.h:219:1:" :/Users/Yama an/ ppData/Roaming/Code/User/globalStorage/bmd.stm32-for-vscode/@xsorry, unimplemented: pa k-dev-tools/arm-none-eabi-gcc/12.2.1-1.2.1/.conThumb-1 'ten /bin/arm-none-eabi-gcc" -c  -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DSTM32U575xhard-floatx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -ICore/Inc -IDrivers/CMSIS/Device/ST/STM32U5xx/Include 
-IDrivers/CM' VFP ABI
  219 | SIS Include -IDrivers/STM32U5xx_HAL_Driver/Inc -IDr{In file included from iv rs/STM32U5xx_HAL_Driver/Inc/Legacy -Og -Wall -fdata-sections -ffunction-se
      | Drivers/CMSIS/Include/cmsis_compiler.h:54 tions -g -gdwarf -^,
                 from ggd
b -Wall -fdata-sectiDrivers/CMSIS/Include/core_cm33.h:206on  -ffunction-sec,
                 from  ions  -MMD -MP -MF"build/stm32u5xx_hDrivers/CMSIS/Device/ST/STM32U5xx/Include/stm32u575xx.h:229al d" -Wa,-a,-ad,-alms=b,
                 from uil /stm32uDrivers/CMSIS/Device/ST/STM32U5xx/Include/stm32u5xx.h:1035 x_ha,
                 from l. st Drivers/STM32U5xx_HAL_Driver/Src/stm32u5xx_hal.c -o build/stm32u5xx_Drivers/STM32U5xx_HAL_Driver/Inc/stm32u5xx_hal_def.h:33h l.o,
                 from
Drivers/STM32U5xx_HAL_Driver/Inc/stm32u5xx_hal_rcc.h:28,
                 from Core/Inc/stm32u5xx_hal_conf.h:265,
                 from Drivers/STM32U5xx_HAL_Driver/Inc/stm32u5xx_hal.h:29,
                 from Core/Inc/main.h:30,
                 from Core/Src/stm32u5xx_it.c:21:
Drivers/CMSIS/Include/cmsis_gcc.h: In function '__get_CONTROL':
Drivers/CMSIS/Include/cmsis_gcc.h:219:1: sorry, unimplemented: Thumb-1 'hard-float' VFP ABI
  219 | {
      | ^
"c:/Users/Yamaan/AppData/Roaming/Code/User/globalStorage/bmd.stm32-for-vscode/@xpack-dev-tools/arm-none-eabi-gcc/12.2.1-1.2.1/.content/bin/arm-none-eabi-gcc" -c  -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DSTM32U575xx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -ICore/Inc -IDrivers/CMSIS/Device/ST/STM32U5xx/Include -IDrivers/CMSIS/Include -IDrivers/STM32U5xx_HAL_Driver/Inc -IDrivers/STM32U5xx_HAL_Driver/Inc/Legacy -Og -Wall -fdata-sections -ffunction-sections -g -gdwarf -ggdb -Wall -fdata-sections -ffunction-secIn file included from ti ns  -MMD -MP -MF"build/sDrivers/CMSIS/Include/cmsis_compiler.h:54tm 2u5xx_hal_adc.d" -Wa,-a,-ad,-alms=build/stm32u5xx_hal_adc.l,
                 from st  rivers/STM32U5xx_HAL_Driver/Src/stmDrivers/CMSIS/Include/core_cm33.h:2063 u5xx_hal_adc.c -o build/stm32u5xx_hal_adc.o
,
                 from Drivers/CMSIS/Device/ST/STM32U5xx/Include/stm32u575xx.h:229,
                 from Drivers/CMSIS/Device/ST/STM32U5xx/Include/stm32u5xx.h:103,
                 from Core/Src/system_stm32u5xx.c:103:
Drivers/CMSIS/Include/cmsis_gcc.h: In function '__get_CONTROL':
Drivers/CMSIS/Include/cmsis_gcc.h:219:1: sorry, unimplemented: Thumb-1 'hard-float' VFP ABI
  219 | {
      | ^"
c:/Users/Yamaan/AppData/Roaming/Code/User/globalStorage/bmd.stm32-for-vscode/@xpack-dev-tools/arm-none-eabi-gcc/12.2.1-1.2.1/.content/bin/arm-none-eabi-gcc" -c  -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DSTM32U575xx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -ICore/Inc -IDrivers/CMSIS/Device/ST/STM32U5xx/Include -IDrivers/CMSIS/Include -IDrivers/STM32U5xx_HAL_Driver/Inc -IDrivers/STM32U5xx_HAL_Driver/Inc/Legacy -Og -Wall -fdata-sections -ffunction-sections -g -gdwarf -ggdb -Wall -fdata-sections -ffunction-secIn file included from ti ns  -MMD -MP -MF"build/stm32u5xx_hal_adc_eDrivers/CMSIS/Include/cmsis_compiler.h:54x d" -Wa,
                 from ,- ,-ad,-alDrivers/CMSIS/Include/core_cm33.h:206ms=build/stm32u5xx_hal_adc_ex,
                 from . st Drivers/STM32U5xx_HAL_Driver/Src/stm32u5xx_hal_adc_ex.c -o build/stm32u5xx_hal_adc_ex.o
Drivers/CMSIS/Device/ST/STM32U5xx/Include/stm32u575xx.h:229,
                 from Drivers/CMSIS/Device/ST/STM32U5xx/Include/stm32u5xx.h:103,
                 from Drivers/STM32U5xx_HAL_Driver/Inc/stm32u5xx_hal_def.h:33,
                 from Drivers/STM32U5xx_HAL_Driver/Inc/stm32u5xx_hal_rcc.h:28,
                 from Core/Inc/stm32u5xx_hal_conf.h:265,
                 from Drivers/STM32U5xx_HAL_Driver/Inc/stm32u5xx_hal.h:29,
                 from Drivers/STM32U5xx_HAL_Driver/Src/stm32u5xx_hal.c:36:
Drivers/CMSIS/Include/cmsis_gcc.h: In function '__get_CONTROL':
Drivers/CMSIS/Include/cmsis_gcc.h:219:1: sorry, unimplemented: Thumb-1 'hard-float' VFP ABI
  219 | {"c
      | :/U ers/Yamaan/AppData/Roaming/Code/User/globalStora^g /bmd.stm3
2-for-vscode/@xpack-dev-tools/arm-none-eabi-gcc/12.2.1-1.2.1/.content/bin/arm-none-eabi-gcc" -c  -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DSTM32U575xx -DUSE_FULL_LL_DRIVER -DUSE_HAL_DRIVER -ICore/Inc -IDrivers/CMSIS/Device/ST/STM32U5xx/Include -IDrIn file included from ive s/CMDrivers/CMSIS/Include/cmsis_compiler.h:54SIS Include -IDrivers/STM32U5xx_HAL_Driver/Inc -IDrivers/STM32U5xx_HAL_Dr,
                 from ive /Inc/LegaDrivers/CMSIS/Include/core_cm33.h:206cy -Og -Wall -fdata-sections -ffunction-sections -g -gdwarf -ggdb -Wall -fdata-sections -ffunction-sec,
                 from t ons  -MDrivers/CMSIS/Device/ST/STM32U5xx/Include/stm32u575xx.h:229M  -MP -MF"build/stm32u5xx_hal_cortex.d" -W,
                 from a,- ,-ad,-alms=build/stm32u5xx_hal_cortex.lst Drivers/STM32U5xx_HAL_DriverDrivers/CMSIS/Device/ST/STM32U5xx/Include/stm32u5xx.h:103/ rc/stm32u5xx_hal_cortex.c -o bu,
                 from il /stm32u5xx_hal_cortex.o
Drivers/STM32U5xx_HAL_Driver/Inc/stm32u5xx_hal_def.h:33,
                 from Drivers/STM32U5xx_HAL_Driver/Inc/stm32u5xx_hal_rcc.h:28,
[... 35 lines cut ...]
      |  AKE.EXE: *** [S^T 32Make.make:224: bui
ld/stm32u5xx_hal_msp.o] Error 1
MAKE.EXE: *** [STM32Make.make:224: build/system_stm32u5xx.o] Error 1
MAKE.EXE: *** [STM32Make.make:224: build/stm32u5xx_it.o] Error 1
MAKE.EXE: *** [STM32Make.make:224: build/stm32u5xx_hal.o] Error 1
In file included from Drivers/CMSIS/Include/cmsis_compiler.h:54,
                 from Drivers/CMSIS/Include/core_cm33.h:206,
                 from Drivers/CMSIS/Device/ST/STM32U5xx/Include/stm32u575xx.h:229,
                 from Drivers/CMSIS/Device/ST/STM32U5xx/Include/stm32u5xx.h:103,
                 from Drivers/STM32U5xx_HAL_Driver/Inc/stm32u5xx_hal_def.h:33,
                 from Drivers/STM32U5xx_HAL_Driver/Inc/stm32u5xx_hal_rcc.h:28,
                 from Core/Inc/stm32u5xx_hal_conf.h:265,
                 from Drivers/STM32U5xx_HAL_Driver/Inc/stm32u5xx_hal.h:29,
                 from Drivers/STM32U5xx_HAL_Driver/Src/stm32u5xx_hal_dma.c:184:
Drivers/CMSIS/Include/cmsis_gcc.h: In function '__get_CONTROL':
Drivers/CMSIS/Include/cmsis_gcc.h:219:1: sorry, unimplemented: Thumb-1 'hard-float' VFP ABI
[... cut ...]
```

I'm not actually sure what this might suggest, I haven't tried to add any external libraries or anything, just a fresh project from CubeMX.  The only work around I've found has been generating a CubeIDE project instead and using the CubeIDE project importer (with generate under root disabled), then things seem to build correctly. (Although I get an annoying `command-line error: language modes specified are incompatibleC/C++(1027)` extension error when I do that but that's a separate issue.

jortbmd, 2023-04-11:
Hi. Thanks for opening up an issue!
I double checked and there seems to be an error in the generated makefile. It does not include the CPU = -mcpu=cortex-m33 flag. If you add cortex-m33 to the cpu variable in the STM32-for-VSCode.config.yaml it should resolve your issue. E.g.: 
```yaml
# MCU settings
targetMCU: stm32u5x
cpu: cortex-m33 # type of cpu e.g. cortex-m4
fpu: fpv4-sp-d16 # Defines how floating points are defined. Can be left empty.
floatAbi: -mfloat-abi=hard
```

yamaan93, 2023-04-14:
thanks for your response. I'll give that a go when I get the chance. 

Also your VSCode extension is a life saver, I absolutely cannot stand cubeIDE but I also depend on the pre-generated files from cubeMX. I appreciate your work

jortbmd, 2023-04-14:
Thank you very much for your kind words! I hope the extension will give you the experience you are looking for.