bmd-studio/stm32-for-vscode #155: Some standard library dependencies not included (C++ build fails)

MucaGinger, 2023-05-13:
I just started using stm32-for-vscode extension. After adding C++ file to the stm32 project and using simple code that uses `iostream` standard library the build fails. I get the following error:

`c:\users\user\appdata\roaming\code\user\globalstorage\bmd.stm32-for-vscode\@xpack-dev-tools\arm-none-eabi-gcc\12.2.1-1.2.1\.content\arm-none-eabi\include\c++\12.2.1\iostream:38:10: fatal error: bits/c++config.h: No such file or directory
   38 | #include <bits/c++config.h>
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.`

I assume there might be an include path to standard libraries missing somewhere? If so, should I manually add it into either of VS Code `.json` files or Makefile? But since I have path for compiler provided inside extension settings, shouldn't everything be okay in terms of standard libraries inclusion?

All the mentioned files are located [here](https://filebin.net/i8kqwxw3cvge5sg2).

Otherwise, a similar or the same issue has been posted under issue #95 however I didn't get any useful information out of it. Except maybe character length of path. But I'm not sure about this either what and how I could solve this issue by trying shortening file path length.

MucaGinger, 2023-05-13:
It seems this is not an stm32-to-vscode issue. I recently realized compiling C++ code using ARM compiler g++ belongs to the "freestanding implementation". Therefore many standard C++ libraries cannot be used because these rely on OS capabilities AFAIK.

MucaGinger, 2023-05-13:
On the other hand I'm not so sure about my claim. Because using any of standard C++ libraries in STM32CubeIDE builds without any errors. So, maybe it is some kind of bug or something similar with this extension?

ARNik, 2023-05-13:
> On the other hand I'm not so sure about my claim. Because using any of standard C++ libraries in STM32CubeIDE builds without any errors. So, maybe it is some kind of bug or something similar with this extension?

Look at the generated code. STM32CubeMX generates different code with CubeIDE and Makefile settings.

MucaGinger, 2023-05-13:
Not sure about the code CubeIDE generates but as for its Makefile, I don't think it is generated when you choose in CubeMX to generate code for CubeIDE.

Otherwise, if CubeIDE allows usage of STL in C++, does that mean standard arrays and vectors may be used when compiling with ARM compiler? I'm lost a bit, I'm not sure anymore, why the Makefile settings are different when you generate code in CubeMX as Makefile project (required by the stm32-for-vscode extension). And whether including `c++configh.h` is this extension's bug or not... any hints very much appreciated.

jortbmd, 2023-06-12:
Hi! Thanks for opening up a new issue for this. As far as I can tell this should work. Unfortunately I cannot open your files any more. Could you perhaps share them again so I can have a look (will have a look at it this week). It might be that you need to add some configuration options for it all to work e.g. using nosys instead of nano specification and adding the c++ standard library.

jortbmd, 2024-04-28:
Closing due to inactivity. Should you have more questions or issues please feel free to re-open this issue again or create a new one.

closed 2024-04-28 by jortbmd