bmd-studio/stm32-for-vscode #156: How do you regenerate a makefile from `STM32-for-VSCode.config.yaml`

Kreijstal, 2023-05-18:
I have modified this yaml, when I build however it reuses the old makefile... how do you fix it?

Kreijstal, 2023-05-18:
Alternatively, why there is no terminal command to generate the makefile, from the yaml configuration without vscode? Could this be done? I mean you can do it with Nodejs can't you?

jortbmd, 2023-06-12:
Hi! Thank you for opening up an issue. If you use the build command the yaml should regenerate the STM32Make.make file (if it needs to be updated). The Makefile generated by CubeMX should stay the same. If the STM32Make.make file does not update please let me know as this might suggest a bug and if so please let me know what the change is in the STM32-for-VSCode.config.yaml and what the current STM32Make.make file is.

There is no terminal command as of yet as currently a lot of the file handling and more stuff is deeply integrated into the extension itself. Extracting some of the logic for STM32 for VSCode to the command line might be something I am going to do more in the future for CI/CD purposes however this might take some time.