bmd-studio/stm32-for-vscode #157: use .env files

Kreijstal, 2023-05-18:
here https://github.com/bmd-studio/stm32-for-vscode/blob/b67fec144d859c20e7ce5e968a0b6146f3fe46bd/src/CreateMakefile.ts#L329 we can see that we hardcode system path on the makefile, it'd be nice to use an env file, which stm32-for-vscode will write, then the makefile reads from the envfile, alternatively, the user variables should be at the top of the file which the user can edit.

Kreijstal, 2023-05-18:
An example `.env` file might look like this:

```bash
GCC_PATH=/usr/bin/
OPENOCD_PATH=/usr/bin/openocd
```

Then in your Makefile you can use `include .env`.

```makefile
include .env

all:
    @echo $(GCC_PATH)
    @echo $(OPENOCD_PATH)
    @echo $(OPENOCD_SCRIPTS)
```

This will include the variables defined in the `.env` file in the environment for the Makefile.

ARNik, 2023-05-20:
I would like to see small strings in my build log either.

Instead of this:
c:/users/myuser/appdata/roaming/code/user/globalstorage/bmd.stm32-for-vscode/@xpack-dev-tools/arm-none-eabi-gcc/12.2.1-1.2.1/.content/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld.exe *** many options ***

Something like:
ld.exe *** many options ***

jortbmd, 2023-06-12:
thanks for opening up an issue for this. Yes you are absolutely right this would make it nicer. It is one of the small things I wanted to fix in a somewhat bigger build overhaul. However this overhaul is taking more time than anticipated. This feature might actually be small enough to integrate rather quickly. Will have a look at it and come back to you on this.

For the smaller string in the build log. I do not know if this is possible as the full path to the tooling is used and I have no desire to make aliases for all the tooling at this point. If somebody knows how this might be done then I am all ears however I do not know of a feature of make that might be able to do it.

jortbmd, 2023-10-19:
Just to give a quick little update. I have been working on this and it is included in the current beta version for the new release. There is still some rough edges for the new release that need to be ironed out, but it is probably useable in most use cases. The environment file is called .stm32env and should be generated once on a new build. If you are in for checking out the beta please find it here: https://github.com/bmd-studio/stm32-for-vscode/releases/tag/v3.2.6-betav1. If you have any feedback or comments they are more than welcome.

jortbmd, 2025-01-20:
I am closing this issue as it is in the newest version: v3.2.10.

closed 2025-01-20 by jortbmd