bmd-studio/stm32-for-vscode #160: make: *** [build/Test_F407.elf] Error 1 (undefined reference to `_exit')

AlexArGC, 2023-07-03:
Any ideas! Through the make is going without errors.

```
arm-none-eabi-g++ build/main.o build/crc.o build/gpio.o build/i2c.o build/spi.o build/stm32f4xx_hal_msp.o build/stm32f4xx_it.o build/system_stm32f4xx.o build/tim.o build/usart.o build/stm32f4xx_hal.o build/stm32f4xx_hal_cortex.o build/stm32f4xx_hal_crc.o build/stm32f4xx_hal_dma.o build/stm32f4xx_hal_dma_ex.o build/stm32f4xx_hal_exti.o build/stm32f4xx_hal_flash.o build/stm32f4xx_hal_flash_ex.o build/stm32f4xx_hal_flash_ramfunc.o build/stm32f4xx_hal_gpio.o build/stm32f4xx_hal_i2c.o build/stm32f4xx_hal_i2c_ex.o build/stm32f4xx_hal_pwr.o build/stm32f4xx_hal_pwr_ex.o build/stm32f4xx_hal_rcc.o build/stm32f4xx_hal_rcc_ex.o build/stm32f4xx_hal_spi.o build/stm32f4xx_hal_tim.o build/stm32f4xx_hal_tim_ex.o build/stm32f4xx_hal_uart.o  build/startup_stm32f407xx.o -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard
-specs=nano.specs  -TSTM32F407VGTx_FLASH.ld  -lc -lm -lnosys  -Wl,-Map=build/Test_F407.map,--cref -Wl,--gc-sections -o build/Test_F407.elf
c:/program files (x86)/gnu arm embedded toolchain/10 2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: c:/program files (x86)/gnu arm embedded toolchain/10 2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard\libc_nano.a(lib_a-abort.o): in function `abort':
abort.c:(.text.abort+0xa): undefined reference to `_exit'
c:/program files (x86)/gnu arm embedded toolchain/10 2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: c:/program files (x86)/gnu arm embedded toolchain/10 2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard\libc_nano.a(lib_a-signalr.o): in function `_kill_r':
signalr.c:(.text._kill_r+0xe): undefined reference to `_kill'
c:/program files (x86)/gnu arm embedded toolchain/10 2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: c:/program files (x86)/gnu arm embedded toolchain/10 2021.10/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/lib/thumb/v7e-m+fp/hard\libc_nano.a(lib_a-signalr.o): in function `_getpid_r':
signalr.c:(.text._getpid_r+0x0): undefined reference to `_getpid'
collect2.exe: error: ld returned 1 exit status
make: *** [build/Test_F407.elf] Ошибка 1
```

jortbmd, 2023-07-03:
Hi! Thanks for opening up an issue for this.
The linker is returning an error. The undefined referent to _exit, _kill,_getpid is standard for a new version of the arm gcc compiler, however it should not return a linker error. Could you try to reinstall the build tools (arm-none-eabi) this might resolve the issue. If not there might be another issue and if so please let me know.

hoanganh1911, 2023-10-16:
I reinstall the build tools (arm-none-eabi) but it still bug

jortbmd, 2023-10-16:
@hoanganh1911  what kind of bug are you experiencing do you have the output of the build logs in your terminal?

jortbmd, 2024-12-23:
I will close this issue due to inactivity. Should any issues arise or if you want to add to this issue, please re-open this one or submit a new issue.

closed 2024-12-23 by jortbmd