bmd-studio/stm32-for-vscode #161: Error: task type build

MarijnBMD, 2023-08-11:
VS Code gives the following error:

Error: there is no registered task type 'build'. Did you miss installing an extension that provides a corresponding task provider?
Error: there is no registered task type 'build'. Did you miss installing an extension that provides a corresponding task provider?

Probably due to the tasks.json config of STM32 for VScode

pintert3, 2023-08-18:
I'm getting this too.

jortbmd, 2023-08-22:
Hi thanks for opening up an issue @MarijnBMD en for confirming the issue @pintert3. Something changed in relation to the task provider. Will investigate further, however if you build once the task should be available once again.

Will let you know once this is resolved.

jortbmd, 2023-09-22:
As a quick little update I can reproduce the bug and the issue is linked to: issue #164  as well. This will be rectified in the next release.

pipi-1, 2023-12-13:
I also faced this problem.

frankqwu, 2024-07-12:
I still have this issue in v3.2.6