bmd-studio/stm32-for-vscode #162: The term 'X86' is not recognized

uldispo, 2023-08-12:
Executing task: "C:\PROGRAM FILES (X86)\ARM GNU TOOLCHAIN ARM-NONE-EABI\12.2 MPACBTI-REL1\BIN\MAKE.EXE" -j16 -f STM32Make.make  

X86: The term 'X86' is not recognized as a name of a cmdlet, function, script file, or executable program.
Check the spelling of the name, or if a path was included, verify that the path is correct and try again.
What to do?

jortbmd, 2023-08-22:
Hi thanks for opening up an issue. As far as I can tell the issue is because of the space in the path name. I will look into this. For now what should work is if you run the install build tools from the extension. 
The other option is escaping the spaces in the arm-none-eabi toolchain path in your settings.

uldispo, 2023-08-28:
Resolved. Use folders without spaces.