bmd-studio/stm32-for-vscode #165: How to make a release build?

ARNik, 2023-10-17:
Hello!

I need to build my project with ```DEBUG = 0``` variable.
Is there a way to set it in the yaml file?

jortbmd, 2023-10-19:
Hi Arnik,

Thanks for opening up a issue once again. This has been on my todo list for a while. I haven't gotten around to it for the full release but I published a beta version which should work in most use cases. The release can be found here: https://github.com/bmd-studio/stm32-for-vscode/releases/tag/v3.2.6-betav1. Please let me know if you find any issues in this release. I hope this works for you. If not don't hesitate to give me some feedback.

jortbmd, 2025-01-20:
Closing this issue as it has been introduced in v3.2.10. Should you want any more features added or if you find any problems do not hesitate te open up another issue.

closed 2025-01-20 by jortbmd