bmd-studio/stm32-for-vscode #166: Unable to import

danielsalyi, 2023-11-05:
1. I am running the import command from the extension 
or the command >st project
2. import the freshly generated project with cubemx (stm32f767zi board)

> Activating environment at "c:\Users\danis\Repositories\nucleoboard\blinky\vcpkg-configuration.json".
> Running vcpkg with arguments ["activate","--project","c:\\Users\\danis\\Repositories\\nucleoboard\\blinky","--json","C:\\Users\\danis\\AppData\\Local\\Temp\\cbff46151332ccc26137bb8219ab75f7"].
warning: vcpkg-artifacts is experimental and may change at any time.
Downloading vcpkg-artifacts bundle 2023-03-14...
Downloading https://github.com/microsoft/vcpkg-tool/releases/download/2023-03-14/vcpkg-ce.tgz
node:internal/modules/cjs/loader:1141
  throw err;
  ^

Error: Cannot find module 'C:\Users\danis\AppData\Local\vcpkg\artifacts-2023-03-14\node_modules\vcpkg-ce'
    at Module._resolveFilename (node:internal/modules/cjs/loader:1138:15)
    at Module._load (node:internal/modules/cjs/loader:979:27)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:122:12)
    at node:internal/main/run_main_module:28:49 {
  code: 'MODULE_NOT_FOUND',
  requireStack: []
}

Node.js v21.0.0
> Failed with exit code "1".
> Failed to activate environment at "c:\Users\danis\Repositories\nucleoboard\blinky\vcpkg-configuration.json".

3. i also closed VScode, open powershell from the root folder, vcpkg activate, i get an error "error: no postscript file: rerun with the vcpkg shell function rather than executable"
4. code . in same cmd or powershell
5. reimport
6. fail again

I spent half my day on this and I am stuck. I dont think its at my end anymore, please help

jortbmd, 2023-11-06:
Hi. Thanks for opening up an issue! I do have the feeling that you might be mixing up the STM32 for VSCode extension with the official STMicroelectronics extension or another extension providing some import functionality. Currently vcpkg is not used in the STM32 for VSCode extension.
You can double check all the command for STM32 for VSCode are prefaced with STM32:.
For the official ST extension they are prefaced with: STM32 VS Code Extension:.

If I am mistaken however and it is an issue with my extension please let me know!

danielsalyi, 2023-11-09:
Yes, you are right, sorry. However I clicked on the link from the extension's site. and i got here. Weird

closed 2023-11-09 by danielsalyi

jortbmd, 2023-11-10:
> Yes, you are right, sorry. However I clicked on the link from the extension's site. and i got here. Weird

No worries. The naming is quite similar. I hope you can get it resolved somehow. And otherwise try STM32 for VSCode as it is designed to work with CubeMX