bmd-studio/stm32-for-vscode #168: Enhancement: Add a build analyzer to get ram&flash usage

Kndy666, 2023-11-17:
Hello!
In some IDEs, there is a build analyzer which can show RAM and FLASH usage, such as CudeIDE.It really help get aware of how much RAM remaining so that I can better optimize my code.
I have wrote a powershell script can automatically get the maxium size of RAM & FLASH and calculate the usage percent. But I don't know how to add it into the extension. Currently, I just use tasks.json to configure vscode. If it can be added into makefile, than it can use `DEBUG` switch to specify the path of .elf file.
Here is my script:
```
function Format-Size() {
    [CmdletBinding()]
    param(
        [Parameter(Mandatory = $true, ValueFromPipeline = $true)]
        [double]$SizeInBytes
    )
    switch ([math]::Max($SizeInBytes, 0)) {
        {$_ -ge 1PB} {"{0:N2}PB" -f ($SizeInBytes / 1PB); break}
        {$_ -ge 1TB} {"{0:N2}TB" -f ($SizeInBytes / 1TB); break}
        {$_ -ge 1GB} {"{0:N2}GB" -f ($SizeInBytes / 1GB); break}
        {$_ -ge 1MB} {"{0:N2}MB" -f ($SizeInBytes / 1MB); break}
        {$_ -ge 1KB} {"{0:N2}KB" -f ($SizeInBytes / 1KB); break}
        default {"$SizeInBytes Bytes"}
    }
}

$Path = (Get-Content .\.stm32env | Select-String -Pattern "ARM_GCC_PATH = ").ToString().Split()[2]
$Target = ".\build\release\Test.elf"
$pinfo = New-Object System.Diagnostics.ProcessStartInfo
$pinfo.FileName = $Path + "\arm-none-eabi-size.exe"
$pinfo.RedirectStandardError = $true
$pinfo.RedirectStandardOutput = $true
$pinfo.UseShellExecute = $false
$pinfo.Arguments = $Target
$p = New-Object System.Diagnostics.Process
$p.StartInfo = $pinfo
$p.Start() | Out-Null
$p.WaitForExit()
$stdout = $p.StandardOutput.ReadToEnd()
$split = $stdout.Split(' ', [StringSplitOptions]::RemoveEmptyEntries)
$text = $split[5] -as [int]
$data = $split[6] -as [int]
$bss = $split[7] -as [int]
$dec = $split[8] -as [int]
$RAM = Format-Size($bss + $dec)
$Flash = Format-Size($text + $data)
$maxRAM = (Get-Content -Path ".\*.ld" | Select-String -Pattern "ORIGIN = " -CaseSensitive)[0].ToString().Split()[13] + 'B'
$maxFlash = (Get-Content -Path ".\*.ld" | Select-String -Pattern "ORIGIN = " -CaseSensitive)[1].ToString().Split()[13] + 'B'

$OutPutInfo = "       {0, 35}{1, 35}" -f "Flash(.text + .data): $Flash", "RAM(.bss + .dec): $RAM"
$OutPut = "Usage: {0, 35}{1, 35}" -f (($Flash / $maxFlash * 100).ToString() + '%'), (($RAM / $maxRAM * 100).ToString() + '%')
Write-Host $OutPutInfo
Write-Host $OutPut
```

lsomaini, 2023-11-22:
Hi, 
I agree with @Kndy666 
Given that most applications are bound by RAM and FLASH memory limitations, having a way to visibly monitor or configure constraints on RAM and FLASH usage would be very beneficial. This feature would enable early detection of any unexpected increases in memory usage, allowing for effective resolution during the initial stages of development. Such proactive monitoring could greatly enhance efficiency and ensure adherence to memory constraints.
That would be a great feature!
Thank you!

jortbmd, 2023-11-22:
Hi All,

Thanks for chiming in. I will look into to this and if time permitted I will see if I can include it in the next release. I can see the value in such a feature. @Kndy666  thanks for including the script this helps a lot when implementing a feature.

JohannesRgnr, 2024-01-02:
In the LinkerFlags section of the STM32-for-VSCode.yaml, I generally add  
- -Wl,--print-memory-usage 
A real build analyzer would be great, though!

jortbmd, 2024-01-09:
@JohannesRgnr that is a good idea! For now I have added it as a default flag in the new release (out now). Will keep this issue open and I will ruminate on this. Might be nice to do some more memory tools like map file interpretation.