bmd-studio/stm32-for-vscode #170: Cortex-Debug: No access to cortex-live-watch variables, registers and Memory

95lux, 2024-01-17:
I can not access the above data from cortex-debug, is this normal behavior? My project builds fine, and i can set breakpoints, add variable watch and debug as normal. But cortex-live-watch variables and the registers tab in the debugging window do not work. 
[image]
I am using an STM32 Nucleo F401RE, with a super bare-bones project setup.

jortbmd, 2024-01-22:
Hi! Thanks for opening up a issue for this. The live watch functionality is not something I have enabled by default in STM32 for VSCode. It is possible to add it however by setting the livewatch configuration in your launch.json file. E.g.: 
```
      "liveWatch": {
        "enabled": true,
        "samplesPerSecond": 4
      },
```
If you had this too your configuration it should work.

For the register/peripherals. These sometimes do not work and I have not been able to debug as to why as of yet. Will try to dive into that a little bit more once I have time available. There is however a xperipherals section which should give you the same information.
Hope this helps!

shanemmattner, 2024-02-08:
Thank you @jortbmd.  I got Live Watch working.  Some of the drop-downs for structs will be delayed or not drop down at all, but stopping execution and clicking the drop down button will show all fields as expected.

Great tool!

jortbmd, 2025-01-20:
Closing this as it seems to be resolved. The watching of xperipeherals and peripherals is still something that needs to be done. For this I have created a new issue: #221.

closed 2025-01-20 by jortbmd