bmd-studio/stm32-for-vscode #172: Add support for STM32C0X

inflowmini, 2024-03-04:
This might not be an extension issue but rather an STMCube MX and OpenOCD issue.

When generating code in MX for STM32C0116-DK, the makefile does not generate correctly and propagates the errors to the .yaml file. Fixing the makefile and the .yaml file makes it possible to correctly compile the code. Unfortunately when flashing, an error is produced by openOCD (Warn : Cannot identify target as an STM32G0/G4/L4/L4+/L5/U5/WB/WL family device.) through the extension.  I assume that this is due to OpenOCD not supporting this particular device. 

I can see that OpenOCD does support this device (via stm32c0x.cfg) but I can't get the extension to flash. STM32Cube IDE can build, compile, and flash the code correctly. 

I'm have tried changing programmers through the extension to no avail. Mostly posting here because I can't find any info on how to fix this.

jortbmd, 2024-03-05:
Hi. Thanks for opening up an issue! I like having these kinds of issues, because in the worst case there is a documented way to work around something. In the best case it is an issue I need to account for in the extension. 
Having said that the most likely fix for this would be to reinstall all the build tools. You probably have an outdated version of openOCD installed. If this does not work please post the exact error from your terminal and I will have a look at it.

inflowmini, 2024-03-05:
You are correct. I had to remove the the build tools and reinstall them. I assumed I had the latest openOCD since I saw version 12 in the build tools but I guess it was outdated.

closed 2024-03-05 by inflowmini

jortbmd, 2024-03-06:
Good to hear it is solved! Should you have any other issues or feature requests feel free to open up a new issues for this.