bmd-studio/stm32-for-vscode #174: Request: add auto-find project in a subfolder

DeflateAwning, 2024-03-11:
Generally, our repo structure is setup as the following, for example:

```
├── firmware
│   ├── Core
│   ├── .cproject
│   ├── project_name.ioc
│   ├── Drivers
│   ├── Middlewares
│   ├── .mxproject
│   ├── .project
│   ├── STM32L4R5ZITX_FLASH.ld
│   └── STM32L4R5ZITX_RAM.ld
├── .git
│   ├── ... git directory
├── .gitattributes
├── .gitignore
├── LICENSE
├── README.md
└── .vscode
    └── settings.json
```

We open VSCode with the workspace directory and the git directory as the same folder.

However, it appears that this extension only works if the `firmware` folder from the previous example is directly in the root of the repo/workspace. It would be nice if it could recursively search the directory to find the `firmware` folder as the project folder.

jortbmd, 2024-03-11:
Hi! Thanks for opening up an issue. This sounds like an easy enough fix. I figured I already had this implemented but apparently not. Will look into this and this will most likely will be part of the next release.

DeflateAwning, 2024-03-12:
It appears that there isn't really support for having the STM32 project in a subfolder at all, at the moment (let alone the original auto-discovery inquiry). 

I've traced it back to the `src/BuildTask.ts` file, where the `build` task is called from the root of the workspace. I believe that the build task simply needs to be called from within the directory in which the yaml config resides, and it should work (as all Makefiles, etc. are setup to be relative from the root of the "project", and not the root of the repo/workspace).

I'm sure there are places other than the build task that would need to be updated, although perhaps updating it in the `executeTask(...)` function definition would solve it everywhere.