bmd-studio/stm32-for-vscode #176: Bug: the yaml "target" field is blank after the setup wizard; add a prompt for the "project name"

DeflateAwning, 2024-03-11:
After running the setup wizard, the `target` field in the configuration yaml is blank. The `target` field is supposed to contain the "project name", per the comment in the yaml file.

The solution is simple: add a prompt in the setup wizard to enter the "project name" (or alternatively, extract it manually).

This fix is important because the error message generated by the build system if the yaml file's `target` is blank requires searching through many layers of makefiles/build files to determine the issue.

jortbmd, 2024-03-11:
Thanks for catching that one! I already have something similar in place for empty or incomplete projects. Will include this check and feature in the next release.

jortbmd, 2025-01-20:
I am closing this issue as it was added in version v.3.2.7 last year.  If there are still issues with it please feel free to re-open the issue or create a new one.

closed 2025-01-20 by jortbmd