bmd-studio/stm32-for-vscode #178: Bug report: Getting "Error: no source files in .project file found" during CubeIDE import

DeflateAwning, 2024-03-12:
I'm trying to import a CubeIDE project, made with the latest version of CubeIDE (v14.1). The `.project` exists and can be found, but it doesn't have any source files listed in it. Apparently that's an issue for this plugin to import.

manmilani, 2025-01-09:
I get the same error. The thing is that CubeIDE (and the old SW4STM32) does not list the source files in the `.project `file. The source files and all project configurations are in the `.cproject` (dot-see-project :D ) file. So this plugin is seemingly looking at the wrong file to begin with.

Dannykcw, 2025-02-07:
This is a duplicate of #102, and it's been two years since then. It might be worth looking into a PR or something.