bmd-studio/stm32-for-vscode #179: Build operation forcefully inserts workspace configuration variables

DeflateAwning, 2024-03-12:
### The Issue
I've notices that when I build, the following configuration variables get added to my VSCode workspace configuration:
* `"cortex-debug.openocdPath"`
* `"cortex-debug.armToolchainPath"`
* Make path too, I think

### Why it's bad
These configuration variables are computer specific, and would be great as global system configuration variables. However, they are getting forcefully inserted into the `.vscode/settings.json` file (or into the `*.code-workspace` file, if I'm using that), which means that file must either be discarded before git committing, or `.gitignore`'d. 

### Code sections
These configuration variables are set from the `src/configuration/cortexDebugConfig.ts` file. 

### Proposed Fix
I'm not too sure what the best fix here is. 

* Might be worth trying to figure out what other extensions do.
* One option would be to just not store the configuration values. It doesn't seem very slow/expensive to locate the files, so maybe they don't need to be stored in the configuration files and can just be located each time.
* Maybe there's a way to make these "User" configurations instead of "Workspace" configurations?

I'm a huge, huge fan of this project, and am completely willing to help improve it by trying to fix some of the issues I find! Thanks for bearing with my suggestions :)

jortbmd, 2024-03-13:
Thanks for opening up this issue. The settings should not be added to the specific workspace. It should very much be a more user specific thing. I feel that this might be a bug due to the way the settings are handled and stored. Will look into this as this has come up more than once, which resulted in some other undesired behaviour.

DeflateAwning, 2024-07-06:
This issue is pretty big for us as it prohibits storing the `settings.json` file in version control (which is needed for other settings). Is there any chance you'd be able to look into it, or help guide me toward where to investigate?

jortbmd, 2024-07-08:
Hi. Thansk for responding to this issue. I have tried to solve this in the 3.2.7 release, which is not ready as of yet due to some issue. But maybe you can check if the current beta solves this issue. Might strip some of the offending features in 3.2.7 and do a smaller release to at least get rid of some of the smaller issues.
You can find the beta here: https://github.com/bmd-studio/stm32-for-vscode/releases/tag/v3.2.7-beta

DeflateAwning, 2024-07-08:
Very exciting - I didn't see the pre-release.

I'll give it a try. I'd be in support of a smaller release if some of the other features aren't ready!

DeflateAwning, 2024-07-14:
Not fixed in `3.2.7-beta`, unfortunately. 

Reproduction:
1. Install `3.2.7-beta` from the `.vsix` file.
2. Find a project compatible with this build system, and create a `.vscode/settings.json` file. Set a random config variable in it (e.g., `{"editor.rulers": [100]}` ), with nothing else in the file.
3. Do a clean build of the project.
4. See that `"cortex-debug.armToolchainPath"` and `"cortex-debug.openocdPath"` paths appear in the file.

jortbmd, 2024-07-29:
Thanks for checking! Will have a look at it, should probably be a small fix. Will update you once it is done. Do note that it might take a little while.