bmd-studio/stm32-for-vscode #180: BUG: SVD files not working anymore, due to svd url not existin anymore.

jortbmd, 2024-03-13:
The orignal source for the SVD files has dissapeared for the STMicro SVD files. Should look for an alternative and implement this once again.

Peter5d5, 2024-03-25:
All the SVD files are also present in the STMCubeMX application under 
`C:\Program Files\STMicroelectronics\STM32Cube\STM32CubeProgrammer\SVD`
How can I add this to the extention?

ARNik, 2024-03-25:
Add svd file to your project and set it in your launch.json configuration. 
for example "svdFile": "STM32G070.svd"

Peter5d5, 2024-03-25:
Ah, my question was not clear enough but thank you for your reply.
I know how to fix my problem but, as jortbmd wrote, bmd-studio normally can find the files but the link is broken.
My suggestion is to look for the SVD files on the local machine.

jortbmd, 2024-03-26:
Thanks @Peter5d5. I prefer not to implement it this way as I cannot guarantee that certain programs are present (e.g. CubeProgrammer.) However I will take it into account when looking for a solution. If I cannot find a better solutions this might be a nice fallback strategy.

Peter5d5, 2024-03-26:
@jortbmd  My mistake, I though it was part of the STM32MXCubeMX package but you are correct. It came with the CubeProgrammer (like the path I provided 🤦)