bmd-studio/stm32-for-vscode #181: BUG: .S assembly files not being compiled

Maobuff, 2024-04-16:
Platform: Windows 10/Linux
stm32-for-vscode v3.2.6
Way to reproduce: 
1) Create a project with STM32U595ZJT6
2) Enable threadx in Middleware
3) Try to build project.
4) *** No rule to make target 'build/tx_initialize_low_level.o', needed by 'build/xxx.elf'.  Stop.

Output of make -f STM32Make.make -qp
```
....
# makefile (from 'STM32Make.make', line 600)
ASM_SOURCES = Core/Src/tx_initialize_low_level.S startup_stm32u595xx.s
....
# Not a target:
build/tx_initialize_low_level.o:
#  Implicit rule search has not been done.
#  Modification time never checked.
#  File has not been updated.
....
build/xxx.elf: .... tx_intialize_low_level.o ....
....
```

The original makefile from CubeMX has ASMM_SOURCES which stm32-for-vscode is completely ignoring.

jortbmd, 2024-04-24:
Hi. Thanks for opening up an issue. That is an interesting one. I have been trying to compile threadx for a hobby project a while ago and ran into similar issues. Can't promise I will have a fix very soon, however will put it on top of my list. If you have any idea of a quick fix in mind I would be happy to hear it.

Maobuff, 2024-04-26:
Splitting assembly code and assembly language files is a good idea (.s and .S). Like the original Makefile from STM32CubeMX: ASM_SOURCES are used for .s and ASMM_SOURCES are used for .S. Main issue right now is that threadx uses a lot of .S files from the threadx library.

I will do some research over the weekend to find out more.

Maobuff, 2024-04-27:
I managed to modify STM32Make.make to build firmware. Here are the steps that I did in order to build.
1) Append ASMM_SOURCES variable from STM32CubeMX Makefile to an ASM_SOURCES of STM32Make.make
2) change `vpath %.s $(sort $(dir $(ASM_SOURCES)))` to `vpath %.s $(sort $(dir $(LOWER_CASE_ASM_SOURCES)))`
3) add `vpath %.S $(sort $(dir $(UPPER_CASE_ASM_SOURCES)))` right below step 2.

Also, STM32Make.make doesn't set AS_DEFS the same as in the original Makefile.

jortbmd, 2024-04-28:
As stated in the pull request, which I thank you for tremendously even though it is not merged. I really appreciate people trying to merge in new stuff. So thank you a bunch for your contribution! A new beta version is out which includes some of your fixed described above. If you would like to test it it can be found here: https://github.com/bmd-studio/stm32-for-vscode/releases/tag/v3.2.7-beta
Could you check if this works for you? You can install it by downloading the VSIX package and then use ctrl/CMD+shift+p and selecting Extensions: Install from VSIX.
Hoping to release a new minor version soon which should fix this for you. Will update you once I have done this.

jortbmd, 2025-01-20:
This issue should be fixed in the newest version of STM32 for VSCode. For this reason I am closing this issue. If the issue still persist please re-open the issue again.

closed 2025-01-20 by jortbmd