bmd-studio/stm32-for-vscode #184: Cmakelist build

danielsalyi, 2024-05-03:
Hello,

Is there a way to build with cmakelist and not from a makefile? Or is there a way to add custom commands to the sidebar?

Tnx!

jortbmd, 2024-05-06:
Hi! Thanks for opening up an issue. 
No not as of yet. Probably the ST extension would work better for the makefile stuff and there is a CMake extension as well which works pretty well.
For now there is no option to add custom commands to the sidebar, however it would be nice to allow some custom makefile commands to be there. Will leave this issue open and work on it at some later date.

DeflateAwning, 2024-07-20:
CubeMX just added support for CMake officially, so this would be a good feature eventually!

danielsalyi, 2024-07-21:
Finally! Thanks for the heads up, I would have missed it