bmd-studio/stm32-for-vscode #187: Compiling project with this extension produces 75KB file; CubeIDE produces 50KB file

yusufso, 2024-05-16:
When I compile my project on STM32CubeIDE with optimization set for size, it produces a 50KB .bin file. 

When doing the same with the stm32-for-vscode extension (still optimizing for size using "-Os" in STM32_for_VSCode.config.yaml), it produces a 75KB file.

What do you think is causing this discrepancy? Aren't both environments using the same toolchain to compile?

I have flashed the 75KB file to a MCU and it does work the way it is supposed to, just find it odd why there's a size difference.

jortbmd, 2024-05-17:
Hi. Thanks for opening up an issue! It should use a similar toolchain. I do not know if there are some slight differences between the two do you maybe have a project you can share which has a similar behaviour? This way I can check what the differences are.

jortbmd, 2024-05-22:
Thanks for sharing. I had a quick look at it and I do not see any significant difference in the makefile in terms of linking. For the STM32 for VSCode one the -lnosys library is included, which you maybe could forgo. On top of that I cannot see how the object files in the CubeIDE makefile are generated. Do you maybe have any optimisations enabled or removed things like debugging symbols?

yusufso, 2024-05-23:
Optimization is set for size (-Os) on both. Debugging symbols are unchanged. I used the extension's importer tool on the project and did not make any further changes.

 I removed the -inosys library from the config.yaml, build size is still 75 kb. 

I downloaded a standalone toolchain which uses the same arm-none-eabi-gcc compiler and the build size was around 50 kb. It seems there must be some setting in the extension that cannot be changed in the config.yaml?