bmd-studio/stm32-for-vscode #188: Project has no STM32-for-VSCode.config.yaml

sidprice, 2024-05-17:
I created a very simple Blinky project using the VSCode STM32 Extension, it builds and runs as expected. I wish to do some Semihosting testing using this project and I need to make some additions to the project to do this. After reading the information in the Visual Studio Code Marketplace for the extension I tried to find the STM32-for-VSCode.config.yaml to learn how to make the changes. I have searched the project folder structure, but that file is nowhere to be found.

The project was generated using CubeMX and used the CMake output format.

Where is this file, or how can I make the changes needed to the linker commands to allow adding, e.g. rdimon.
Sid

jortbmd, 2024-05-20:
Hi. Thanks for opening up an isssue. Hearing your issue I have the idea that you are using the official ST extension and not STM32 for VSCode. Could you double check? For the STM32-for-VSCode.yaml to show up you need to select the a makefile project.

sidprice, 2024-05-20:
I am so sorry, you are correct, thank you for taking the time to reply.

closed 2024-05-20 by sidprice