bmd-studio/stm32-for-vscode #190: Arm toolchain not detected: folder just downloaded

darderik, 2024-06-04:
Hello, while I didn't find any issues at all on my laptop and succesfully configured the extension,i'm unable to do it on my desktop pc. Even after deleting multiple times the bdm folder in AppData/Code, thus reinstalling the build tools, the extension  can't find the arm toolchain folder even if it's there. Did I do something wrong on my end? Thank you!

[image]

Saksham-P, 2024-06-05:
I am also experiencing the same issue. I have tried a few things:
1. I tried reinstalling the STM32 for VSCode Extension which didn't really do anything.
2. After clicking **Install Build Tools**, it successfully installs the ARM Toolchain, Make executable, and openOCD executable, but regardless the extension cannot detect the ARM Toolchain and prompts the user to still Install Build Tools
3. I replaced the variable in `"...\AppData\Roaming\Code\User\settings.json` for the field `stm32-for-vscode.armToolchainPath` from **false** to **"..AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\@xpack-dev-tools\arm-none-eabi-gcc\13.2.1-1.1.1\.content\bin"** and then tried building the project using command palette options, but that also failed
4. I also realized that running the **Check Build Tools** command sets the `stm32-for-vscode.armToolchainPath` field to **false** in the `settings.json` file.

Saksham-P, 2024-06-05:
I found a work around (for Windows, maybe same works for mac).
1. https://developer.arm.com/downloads/-/gnu-rm - Download the `gcc-arm-none-eabi-10.3-2021.10-win32.zip` file and extract it

2. Once extracted, take the 4 folders `bin`, `lib`, `share`, `arm-none-eabi` and drag them to `...\AppData\Roaming\Code\User\globalStorage\bmd.stm32-for-vscode\@xpack-dev-tools\arm-none-eabi-gcc\13.2.1-1.1.1\.content` or wherever your STM32 for VSCode toolchain is downloaded. If it asks whether to replace the files, just say yes. 

3. In the STM32 For VSCode Extension, click `Check Build Tools`, this should update all the paths in settings

There are probably things that are not necessary in the above steps, but it worked for me

darderik, 2024-06-05:
> I found a work around (for Windows, maybe same works for mac).
>
> 1. https://developer.arm.com/downloads/-/gnu-rm - Download the `gcc-arm-none-eabi-10.3-2021.10-win32.zip` file and extract it
>
>
>
> 2. [...]

Thank you so much! Will try it asap

DeflateAwning, 2024-07-06:
Seems like this issue was as a result of one of the required downloads being unavailable at the time, as this issue doesn't appear to have happened again for my team. Perhaps this issue can be closed, unless there's a reproduction which still fails?

jortbmd, 2024-07-08:
Might be good to have a back-up for this, as I want to move away from the current solution I am using (xPack), as they seem to be moving away of providing individual downloads for tooling without using some sorf of configuration file.

darderik, 2025-09-05:
> Thanks for re-opening the issue. The extension should look for the newest folder and does not version pin it. However what might be the case it that new build tools were installed and that the .stm32env variable was not updated. Could you maybe check if you reinstall the build tools again and then delete the .stm32env file in your project if it then works?

My bad. Didn't even realize that the paths were stored in stm32env..
Deleting the stm32env file allows to make a clean install of the toolchain.
I sync my project with my 2 PCs with onedrive so i had the same stm32env file on both. 
Thank you for the help.

nikodemin, 2025-12-26:
I have issues with building project, I got error:
`Something went wrong during the build process: Error: Something went wrong with configuring the workspace. ERROR: Error: Getting the absolute compiler path failed. Please create an issue on the STM32 for VSCode GitHub if you encounter this.`

I am using Fedora 41, and cannot figure out why extension cannot find toolchain (it exists on specified path). Could you help me with this?

settings.json: 
```{
    "cortex-debug.armToolchainPath": "/usr/bin/arm-none-eabi-g++",
    "cortex-debug.openocdPath": "/usr/bin/openocd",
    "stm32-for-vscode.openOCDPath": "/usr/bin/openocd",
    "stm32-for-vscode.armToolchainPath": "/usr/bin/arm-none-eabi-g++"
}
```
.stm32env:
```
# environment variable file used by stm32-for-vscode and the STM32Make.make makefile
# Other environment variables can be added here. If wanting to use the generated makefile in CI/CD context please
# configure the following variables: GCC_PATH, OPENOCD

ARM_GCC_PATH = /usr/bin/arm-none-eabi-g++
OPENOCD = /usr/bin/openocd
 ```

when i press button "Install build tools" something got installed via xpm, but I cannot build project either.