bmd-studio/stm32-for-vscode #193: `make` output on Windows has interwoven stdout/stderr streams

DeflateAwning, 2024-07-20:
It appears that xPack's make on Windows has some serious issues where the stdout and stderr logs get interwoven.

I've opened an issue over on their repo. Hopefully they're able to fix it at the source: https://github.com/xpack-dev-tools/windows-build-tools-xpack/issues/27

However, if can't/don't fix it, it would be nice if you were able to redirect stderr to stdout as you call that `make.exe` command (for Windows only) so that the build outputs are readable on Windows.

Here's an example of the bizzare output:

[image]

DeflateAwning, 2024-07-22:
Update: the issue appears to be present on some Mac systems, and on Windows. One solution/work-around is to force `-j1` to make the output readable (at the cost of build speed).

Can I please request adding a selector or "use multiple threads" checkbox or similar in the extension which lets you pick the number of threads to use?

jortbmd, 2024-07-29:
I am aware of this issue. Unfortunately this is due to an update in make from 4.0 onwards. To counter this you can add the -O flag to make and it should separate the streams.
e.g. add this to the .yaml file:
```yaml
makeFlags:
  - -O  # use this option when the output of make is mixed up only works for make version 4.0 and upwards
```
Hope this helps.

DeflateAwning, 2024-07-29:
Would you accept a PR that checks the `make` version and sets this flag, if appropriate?

jortbmd, 2024-07-30:
Thanks for offering! Yes I would. If this is an universal thing for make 4.0 then it would make sense to have it. Implementation wise it would probably make sense to do this check on extension start-up. The function checkBuildTools in src/buildTools/index.ts is run upon start-up and would probably be the most logical entrypoint for implementing  the feature. The injection of the flag should probably happen in the buildSTM function in src/BuildTask.ts. Hope this helps and I am looking forward to the PR!