bmd-studio/stm32-for-vscode #197: Request: add CONTRIBUTING.md document which outlines how to get started contributing

DeflateAwning, 2024-08-02:
### Info to Include
* Installing dependencies? Which version of node? Is v20 fine?
* How do you run it? Or a link to a [page like this](https://code.visualstudio.com/api/get-started/your-first-extension) for how to start running a dev version of it. Assume that most people developing this extension are career C-language devs who think VS Code is very cool, but who have never used Typescript.
* How to trigger unit tests? Seems like there are custom [undocumented] `npm run xxx` commands.
* What is the `docs/` folder? How is it generated? Is it expected that contributors do/don't regenerate it? Feel like it shouldn't be in the repo, tbh.

jortbmd, 2024-08-05:
Thanks for creating a contribution document! I have merged it. You did an awesome job, so thank you!

closed 2024-08-05 by jortbmd

DeflateAwning, 2024-08-05:
No worries! Appreciate the postivite feedback :)

PR #198 (2024-08-05): Create CONTRIBUTING.md file
Fixes #197