bmd-studio/stm32-for-vscode #200: I found an issue with mixed C and C++ programming.

ghosoft, 2024-08-14:
When doing mixed C and C++ programming, the list of CPP source files in the Makefile gets removed in the generated STM32Make.make.

Makefile:
```
######################################
# source
######################################
# C++ sources
CPP_SOURCES =  tester/png_draw_file.cpp Middlewares/Third_Party/PNGdec/src/PNGdec.cpp

# C sources
C_SOURCES =  \
```

STM32Make.make:
```
tester/util.c

CPP_SOURCES = \

# ASM sources
ASM_SOURCES =  \
```

ghosoft, 2024-08-16:
Thanks for your contribution, your project make our work easier .
Could you pls help to see whether you have any suggestion on this bug ?

ghosoft, 2024-08-19:
thank you

[STM32F407_FSMC.zip](https://github.com/user-attachments/files/16653368/STM32F407_FSMC.zip)

.yaml and makefle is here
version is v3.2.6

jortbmd, 2024-08-19:
Thanks for sharing. I see that the language is set to C at the top of the yaml file. Could you set it to C++? This should resolve this issue. If not please let me know.

jortbmd, 2024-12-23:
I am closing this issue for now due to inactivity. Should anything pop up again or should not be resolved. Please feel free to re-open this issue or create a new one.

closed 2024-12-23 by jortbmd