bmd-studio/stm32-for-vscode #203: File paths which include `#` character break the generated Makefile

DeflateAwning, 2024-09-04:
Shocking I came across this, but file paths which contain a `#` symbol (for example, in the STM32 ST Cryptolib library: `./Middlewares/ST/ST_Cryptolib/Inc/RSA/PKCS#1v15/rsa_pkcs1v15.h`). 

I think it's an easy fix, where the `#` symbol needs to be replaced with `\#`.

Would you accept a fix for just this and publish a release with just this change?

jortbmd, 2024-09-04:
Thanks for noticing these things and the willingness to help. Yes I would accept a fix and publish the release. There have been a couple of hot fixes I have been putting off for a new version but this might be a nice reason to do a tiny version bump in between.

DeflateAwning, 2024-09-04:
Should I branch off of main? Is that where the release will be tagged from?

closed 2024-09-04 by jortbmd

DeflateAwning, 2024-09-04:
Would really appreciate if you could tag a release of this pretty soon! This will be very helpful to us.

jortbmd, 2024-09-05:
Yes. Will run some tests today and if all looks good will release a new minor version.
I was also wondering if you could provide a small project which uses and compiles the crypto project, which fails now and succeeds on your change. This way I can add it to the test suite, to prevent regression of this in the future.

jortbmd, 2024-09-05:
New version 3.2.8 is published!

PR #204 (2024-09-04): Fix hash symbol in include path breaking Makefile gen
Fixes #203