bmd-studio/stm32-for-vscode #205: Build fails  after updating to 3.2.8 version

BlazejKrysztofiak, 2024-09-06:
Hi
My extension automatically updated to version 3.2.8 and my build started to fails:
Here is my STM32Make.make file:
`##########################################################################################################################
# File automatically-generated by STM32forVSCode
##########################################################################################################################

# ------------------------------------------------
# Generic Makefile (based on gcc)
#
# ChangeLog :
#   2024-04-27 - Added env file inclusion. 
#                Added way to overide: build directory, target name and optimisation.
#                Added GCC_PATH by env file to not make the makefile machine dependent.
#                Currently folder structure in build directory is preserved
#                Switching of debug/release build output folder now happens based on debug flag
#   2017-02-10 - Several enhancements + project update mode
#   2015-07-22 - first version
# ------------------------------------------------

######################################
# Environment Variables
######################################
# Imports the environment file in which the compiler and other tooling is set
# for the build machine.
# This can also be used to overwrite some makefile variables
file_exists = $(or $(and $(wildcard $(1)),1),0)
ifeq ($(call file_exists,.stm32env),1)
  include .stm32env
endif

######################################
# Target
######################################
# This is the name of the embedded target which will be build
# The final file name will also have debug or release appended to it.
TARGET ?= ComApp

#######################################
# Build directories
#######################################
# Build path can be overwritten when calling make or setting the environment variable
# in .stm32env

BUILD_DIRECTORY ?= build

######################################
# Optimization
######################################
# Optimization is switched based upon the DEBUG variable. If set to 1
# it will be build in debug mode with the Og optimization flag (optimized for debugging).
# If set to 0 (false) then by default the variable is used in the configuration yaml
# This can also be overwritten using the environment variable or by overwriting it
# by calling make with the OPTIMIZATION variable e.g.:
# make -f STM32Make.make -j 16  OPTIMIZATION=Os

# variable which determines if it is a debug build
DEBUG ?= 1

# debug flags when debug is defined
OPTIMIZATION ?= -O0

RELEASE_DIRECTORY = $(BUILD_DIRECTORY)/debug
ifeq ($(DEBUG),1)
  # Sets debugging optimization -Og and the debug information output
  OPTIMIZATION_FLAGS += -Og -g -gdwarf -ggdb
  $(TARGET) := $(TARGET)-debug
  RELEASE_DIRECTORY := $(BUILD_DIRECTORY)/debug
else
  OPTIMIZATION_FLAGS += $(OPTIMIZATION)
[... 320 lines cut ...]
cmd /c rd /s /q build
 *  Terminal will be reused by tasks, press any key to close it. 

 *  Executing task in folder ComApp: & 'C:\USERS\BLAKRZ\APPDATA\ROAMING\CODE\USER\GLOBALSTORAGE\BMD.STM32-FOR-VSCODE\@XPACK-DEV-TOOLS\WINDOWS-BUILD-TOOLS\4.4.1-2.1\.CONTENT\BIN\MAKE.EXE' -j16 -O -f STM32Make.make  

cmd /e:on /c md build\debug
cmd /e:on /c md build\debug\Drivers\STM32G4xx_HAL_Driver\Src
cmd /e:on /c md build\debug\Src
cmd /e:on /c md build\debug\parentparentShared\Api
cmd /e:on /c md build\debug\parentparentShared\CircularBuffer
cmd /e:on /c md build\debug\parentparentShared\CmdParser
cmd /e:on /c md build\debug\parentparentShared\Crc
cmd /e:on /c md build\debug\parentparentShared\ExternalFlash
cmd /e:on /c md build\debug\parentparentShared\FileSystem
cmd /e:on /c md build\debug\parentparentShared\LittleFs
cmd /e:on /c md build\debug\parentparentShared\Log
cmd /e:on /c md build\debug\parentparentShared\Modem
cmd /e:on /c md build\debug\parentparentShared\Spi
cmd /e:on /c md build\debug\parentparentShared\SwTimer
MAKE.EXE: *** No rule to make target 'build/debug/parentparentShared/Api/Api.o', needed by 'build/ComApp.elf'.  Stop.
MAKE.EXE: *** Waiting for unfinished jobs....
cmd /e:on /c md build\debug\parentparentShared\Uart
cmd /e:on /c md build\debug\parentparentShared\Utils

 *  The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command & 'C:\USERS\BLAKRZ\APPDATA\ROAMING\CODE\USER\GLOBALSTORAGE\BMD.STM32-FOR-VSCODE\@XPACK-DEV-TOOLS\WINDOWS-BUILD-TOOLS\4.4.1-2.1\.CONTENT\BIN\MAKE.EXE' -j16 -O -f STM32Make.make " terminated with exit code: 1.

yuncheng2024, 2024-09-06:
This might help you
https://bit.ly/3yXXzbX
Passcode: d5kp14h
You must have gcc compiler installed

BlazejKrysztofiak, 2024-09-06:
I have gcc compiler installed, when I get back to older version then it works fine.

BlazejKrysztofiak, 2024-09-09:
Yup, 3.2.7 is working fine. Thank you for answering :)

jortbmd, 2025-01-20:
Hi. The newest version is out which might fix your issue. Could you maybe check if version v3.2.10 work for you?

BlazejKrysztofiak, 2025-01-21:
Hi. It works now. Thank you for your work :)

jortbmd, 2025-01-21:
Good to hear! Will close this issue for now. Should any further issues arise feel free to create a new issue or reopen this one.

closed 2025-01-21 by jortbmd