bmd-studio/stm32-for-vscode #206: Build won't generate bin file after updating to 3.2.8

LiangYiJlu, 2024-09-06:
After updating to 3.2.8, there're only elf and map files in build folder. I need bin file for my bootloader, so I reinstalled 3.2.6. Now .bin file is back.

MX-Master, 2024-09-07:
Confirmed

arktrin, 2024-09-10:
so, when this issue will be fixed in release version ?

jortbmd, 2024-09-11:
Hopefully pretty soon. I am still trying to resolve and additional issue which has to do with sources from parent folders and as soon as I got that tackled the new version will be released. If you want to use the new features from the 3.2.8 release I can do a pre-release of the new minor release which addresses this current issue already.

arktrin, 2024-09-13:
will wait, thanks

jortbmd, 2024-09-27:
I have resolved this in the new minor release that will come up. I will still run some tests before releasing it but the beta can be found here:
https://github.com/bmd-studio/stm32-for-vscode/releases/tag/v3.2.9-beta

closed 2025-01-20 by jortbmd