bmd-studio/stm32-for-vscode #207: Flashing fails with new update

NuclearTea, 2024-09-06:
Build succeeds however, doesnt produce the appropriate build files (build folder is basically empty).

Error:

[image]

Line in file: 
```
flash: $(BUILD_DIRECTORY)/$(TARGET).elf
	"$(OPENOCD)" -f ./openocd.cfg -c "program $(BUILD_DIRECTORY)/$(TARGET).elf verify reset exit"

```

DeflateAwning, 2024-09-07:
This is v3.2.8, right? What operating system?

Can you please include a larger section of the error? The part you've included isn't possible to debug from, as _every_ possible failure during the `make flash` command will show that exact message ("Error 1").

I'm unable to reproduce this issue in its current state.

jortbmd, 2024-09-09:
Hi. Thanks for opening up an issue for this. I cannot really tell what the issue might be from the input you have given please provide some more info if available. You can also run the make processing by typing in the command: ``make -f STM32Make.make`` and see what the output is.
It might also be possible that something changed for your .stm32env file. Could you please delete this and check again?

vaibhavkpr, 2024-09-11:
Hi! I would like to add to the issue details. The issue is with the v3.2.8 running on Windows. Here is a clearer screenshot of the flash error I am seeing.
[image]

vaibhavkpr, 2024-09-11:
@jortbmd No problem! Yes so far I have attempted deleting the extension and reinstalling it, reverting back to a previous version (Seems to work after setting the correct version directory of OPENOCD in the .stm32env file, doesn't still fix the issue in the new version despite having the right paths for both OPENOCD & GCC) & deleting the .stm32env file and regenerating it by attempting a new clean build.

It does appear that OPENOCD was updated to a newer version and I did verify that the updated paths exist.

Here is what the .stm32env file looks like:
```
# environment variable file used by stm32-for-vscode and the STM32Make.make makefile
# Other environment variables can be added here. If wanting to use the generated makefile in CI/CD context please
# configure the following variables: GCC_PATH, OPENOCD

ARM_GCC_PATH = C:\Users\USERNAME\APPDATA\ROAMING\CODE\USER\GLOBALSTORAGE\BDM.STM32-FOR-VSCODE\@XPACK-DEV-TOOLS\ARM-NONE-EABI-GCC\13.3.1-1.1.1\.CONTENT\BIN
OPENOCD = C:\USERS\USERNAME\APPDATA\ROAMING\CODE\USER\GLOBALSTORAGE\BMD.STM32-FOR-VSCODE\@XPACK-DEV-TOOLS\OPENOCD\0.12.0-4.1\.CONTENT\BIN\OPENOCD.EXE
```

jortbmd, 2024-09-16:
That is a weird bug. Looks like the path is ok. Can you double check if you run the openocd command with the full path in your terminal with and without quotes. 
e.g. 
``C:\USERS\USERNAME\APPDATA\ROAMING\CODE\USER\GLOBALSTORAGE\BMD.STM32-FOR-VSCODE\@XPACK-DEV-TOOLS\OPENOCD\0.12.0-4.1\.CONTENT\BIN\OPENOCD.EXE``
and 
``"C:\USERS\USERNAME\APPDATA\ROAMING\CODE\USER\GLOBALSTORAGE\BMD.STM32-FOR-VSCODE\@XPACK-DEV-TOOLS\OPENOCD\0.12.0-4.1\.CONTENT\BIN\OPENOCD.EXE"``
to see what happens. Also what shell are you using? CMD, Powershell, fish?

DeflateAwning, 2024-09-17:
Looks like Powershell. 

The `/usr/bin/sh` line on the 10th line seems noteworthy (right where the failure happens from the `flash` command within the Makefile). Why would that appear on a Windows system? Is that perhaps the command that it can't find?

vaibhavkpr, 2024-09-17:
Yes, I'm running PowerShell. I am able to run the OPENOCD command with the full path with and without quotes.

jortbmd, 2024-09-17:
@DeflateAwning I do think your on to something right there. I did a quick inspection of the generated makefile, however it should not add this.
@vaibhavkpr  could you check which version of make you have installed? If you run ``make --version`` it should tell you the version number.

DeflateAwning, 2024-09-17:
From the screenshot with the blue redactions, it's the xpack v4.4.1 make

jortbmd, 2024-09-17:
@DeflateAwning Thanks for pointing that out. Somehow I seem to skip over stuff in the materials provided. So thanks for paying attention.
Seems like this is the newest make out there.
I feel like I need to run some test to check out what is causing this issue. It is one that kind of baffles me at the moment. I cannot think of a way that the ``/usr/bin/sh`` should be injected.
@vaibhavkpr as a check could you maybe add the following to the STM32Make.make makefile:
```
info:
	echo $(OPENOCD)
```
and then run: ``make -f STM32Make.make info``.
This way I can check if it is in the Openocd string or that this originates elsewhere.

vaibhavkpr, 2024-09-17:
@jortbmd I'm running `GNU Make 4.4.1` currently. Upon injecting the echo in the STM32Make.make file here's what I found:
```
make -f STM32Make.make info
echo C:\USERS\Username\APPDATA\ROAMING\CODE\USER\GLOBALSTORAGE\BMD.STM32-FOR-VSCODE\@XPACK-DEV-TOOLS\OPENOCD\0.12.0-4.1\.CONTENT\BIN\OPENOCD.EXE
C:\USERS\Username\APPDATA\ROAMING\CODE\USER\GLOBALSTORAGE\BMD.STM32-FOR-VSCODE\@XPACK-DEV-TOOLS\OPENOCD\0.12.0-4.1\.CONTENT\BIN\OPENOCD.EXE
```

Oddly though when I added the path to `make` in settings.json in VSCode for the integrated terminal (PowerShell), it seems to resolve the issue altogether and run flash without errors.
```
"terminal.integrated.env.windows": {
        "PATH": "C:\\Users\\vaibh\\AppData\\Roaming\\Code\\User\\globalStorage\\bmd.stm32-for-vscode\\@xpack-dev-tools\\windows-build-tools\\4.4.1-2.1\\.content\\bin;C:\\Program Files\\msys64\\usr\\bin;${env:PATH}"
},

DeflateAwning, 2025-01-20:
I believe this issue may have been related to a bad upgrade between two versions. Seems that it hasn't come up again for me at least

NuclearTea, 2025-01-22:
Has not come up for me either in a long time. Might be ok to close this.

jortbmd, 2025-01-23:
Thanks for replying. In this case I will close it for now. Should it happen again feel free to re-open the issue.

closed 2025-01-23 by jortbmd