bmd-studio/stm32-for-vscode #210: make (e=87): The parameter is incorrect.

Nalydmerc, 2024-09-12:
Hello! First of all, thank you for all of your work and supporting this extension. Your work saves me a lot of time during the day.

I'm building a project with LVGL, which is quite a large graphics library. The makefile that the extension has generated is here:

[STM32Make.make.txt](https://github.com/user-attachments/files/16984743/STM32Make.make.txt)

And the extension configuration is here:

[STM32-for-VSCode.config.yaml.txt](https://github.com/user-attachments/files/16984870/STM32-for-VSCode.config.yaml.txt)

When I run a clean build, I get the error

```
make (e=87): The parameter is incorrect.
STM32Make.make:1170: recipe for target 'build/300_400085.elf' failed
MAKE.EXE: *** [build/300_400085.elf] Error 87

 *  The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command & 'C:\PROGRAM FILES\MICROCHIP\MPLABX\V6.00\GNUBINS\GNUWIN32\BIN\MAKE.EXE' -j16 -f STM32Make.make " terminated with exit code: 1. 
```

The full error log is here:

[ErrorLog.txt](https://github.com/user-attachments/files/16984771/ErrorLog.txt)

Just from a cursory search, I found [this thread](https://community.st.com/t5/stm32cubeide-mcus/building-error-from-cubeide-1-14-0/td-p/624782) which suggests that this might be a Windows limitation that the executed command exceeds 32K characters. In any case, I haven't had this problem with 3.2.6, and switching back to that version builds the project just fine.

jortbmd, 2024-09-16:
Thanks for opening up this issue and providing research and a possible cause. I can remember hitting this same issue once myself. I will think about this and see if I can find a nice solution.

jortbmd, 2025-01-20:
I have found a solution for it. It is implemented in the V3.2.10 version of the extension. Could you maybe check if this resolves your issue?