bmd-studio/stm32-for-vscode #212: It's not the same as before Build

ClingYang, 2024-09-24:
[image]
In the past, I could start compiling regardless of whether there was this "build" folder or not, but now I get an error. I am very confused. I don't know why my computer system is Windows 10

jortbmd, 2024-09-24:
Thanks for opening up an issue for this. With this release the makefile moved to a more advanced directory structure. It could be that this is causing some issues. Could you maybe translate the above output to English? It might help me understand the errors a bit better.
I also see that you are using the Mingw32-make. Could you maybe tell me which version you are on? For now should this be an issue which continuous you could rollback the version to 3.2.6 to remedy the problem for now and I will update you once a fix might be available for your issue.

ClingYang, 2024-09-24:
[image]
You're welcome, I'll repeat the problem I encountered: in previous versions, I was able to use "build", but now I can't because "the subdirectory or file build\debug already exists". I don't know if the translation can accurately convey the meaning. I'm reproducing this problem on Windows 10. This is a screenshot of my compilation environment。

qwertym88, 2024-09-26:
@jortbmd I might figure it out. In _STM32Make.make_ mkdir_function. 
```
mkdir_function = mkdir -p $(1)
ifeq ($(OS),Windows_NT)
  convert_to_windows_path = $(strip $(subst /,\,$(patsubst %/,%,$(1))))
  REMOVE_DIRECTORY_COMMAND = cmd /c rd /s /q
  mkdir_function = cmd /e:on /c if not exist $(call convert_to_windows_path,$(1)) md $(call convert_to_windows_path,$(1))
endif
```
When creating new directory, you should check if it already exists before calling _md_, otherwise an error occurs saying that "子目录或文件xxx已存在" (directory/file already exists). The simplest way to solve it is to change 
`mkdir_function = cmd /e:on /c md $(call convert_to_windows_path,$(1))` 
to
`mkdir_function = cmd /e:on /c if not exist $(call convert_to_windows_path,$(1)) md $(call convert_to_windows_path,$(1))`
so that every time _mkdir_function _ is called, it checks if the directory exists first, if not, creates one. This simple patch works fine for me.  

I'm also wondering if there are any ways to modify it manually, something like modifying the source code (where is this line of code generated). I'm struggling with a project and the deadline is approaching, so I would like a quicker way to get this extension work on my vscode (rather than waiting for the next fix).  I appreciate it very much.

@ClingYang 按这样改一下就能用了。然后手动输入 make -j16 -f STM32Make.make 就好，clean/flash/debug 任务同理。和make、arm-toolchain、openocd版本都没关系，纯粹是makefile小问题。

ClingYang, 2024-09-27:
@qwertym88 我退回这个插件版本了

jortbmd, 2024-09-27:
@ClingYang and @ qwertym88 thanks for the input! Will update this accordingly and I will put a small patch version out which addresses it. Will let you know once it is live.

jortbmd, 2024-09-27:
Have created a new version. It is currently in beta because I still need to test it some more. However if you want to test it you can find it here and install it manually: https://github.com/bmd-studio/stm32-for-vscode/releases/tag/v3.2.9-beta

qwertym88, 2024-09-27:
Thanks a lot, it works fine.

eisterman, 2024-10-25:
Hello, I have the same problem. Can I ask if you already know when the v3.2.9 will be officially released?
Thank you very much for all your work!

camerakang, 2024-10-28:
> 已经创建了一个新版本。目前处于测试阶段，因为我还需要对其进行更多测试。但是，如果您想测试它，可以在这里找到它并手动安装：[https ://github.com/bmd-studio/stm32-for-vscode/releases/tag/v3.2.9-beta](https://github.com/bmd-studio/stm32-for-vscode/releases/tag/v3.2.9-beta)

This version still has errors

jortbmd, 2024-10-28:
Hi All. Thanks for responding to this issue. Some more work is required to get the new patch version out, as I feel that the current implementation of the Makefile causes to many side effects so I need to make some time to fix some more issues. Will let you know when this is done and an new test version is available.

jortbmd, 2025-01-20:
Hi all. The newest version is out. Could you please check if the issue still persists for version v3.2.10?